slic3r/Slic3r #2836: Detect thin walls doesn't merge paths

Mikk36, 2015-05-18:
https://dl.dropboxusercontent.com/u/54281287/slic3rThinIssue.zip
As an example from that file, lines 30796 to 30805:
G92 E0
G1 Z5.400 F9000.000
G1 X-16.203 Y37.850 F9000.000
G1 Z5.200 F9000.000
G1 E1.20000 F600.00000
G1 X-28.706 Y30.631 E1.46507 F1800.000
G1 X-28.706 Y30.631 F9000.000
G1 X-16.203 Y37.850 E2.32169 F1800.000
G1 E1.12169 F600.00000
G92 E0

These are erroneous:
G1 X-28.706 Y30.631 F9000.000
G1 X-16.203 Y37.850 E2.32169 F1800.000

This issue is present in the git version, didn't seem to be there with 1.2.6, but can't test it right now (for some reason it started crashing, which is why I'm testing out with git version).

Mikk36, 2015-05-20:
Just as a comparison: the same profile and stl file with 1.2.6-exp:
G92 E0
G1 Z5.400 F9000.000
G1 X-28.724 Y30.621 F9000.000
G1 Z5.200 F9000.000
G1 E1.20000 F600.00000
G1 X-16.185 Y37.860 E1.45625 F1800.000
G1 E0.25625 F600.00000
G92 E0

alexrj, 2015-05-22:
Confirmed. The issue is that gaps are being filled twice:

[image]

```
Making perimeters for layer 33
  0 thin walls detected
  1 gaps filled with extrusion width = 0.93
  1 gaps filled with extrusion width = 0.31
  1 thin walls detected
```

alexrj, 2015-05-22:
Fixed! Thank you.

closed 2015-05-22 by alexrj

a4jp-com, 2015-05-22:
Is it okay to compile the software with the error below? I found it in the latest update.

Building and testing Math-PlanePath-119 ... FAIL
! Installing Math::PlanePath failed. See /home/travis/.cpanm/work/1432286671.2962/build.log for details. Retry with --force to force install it.

xdissent, 2015-05-22:
I just tried updating and got the same failure. It's due to a specific test in Math::PlanePath that is failing on my machine (OS X), possible due to some integer overflow? The test in question deals with `round_up_pow()`, which is a new function in 119 that slic3r doesn't use. There's a check at the top of the test for "modulo 64bit dodginess" so maybe another similar hack is required for our machines? It passes fine in a linux VM. Anyway, since the function in question isn't used in slic3r, I just added a `--no-test` in `Build.PL` for the Math::PlanePath lib as a temporary workaround.