slic3r/Slic3r #2837: what's the g-code for color?

columbus1231, 2015-05-19:
I am wondering how color is represented in g-code after conversion from an amf file. 
For example, just like a straight linear movement is represented in g-code as G1 Xnnn Ynnn Znnn Ennn Fnnn Snnn. How is the red green blue (rgb) color code read in g-code by a 3d printer in a section?

markwal, 2015-05-19:
As far as I know, there isn't such a g-code.  (MachineKit uses M420 to set RGB of the LED illumination using R (rpm) E (extruder) B (b extruder) parameters to map to R G B.)

Color change in a multi-extruder set up using AMF is accomplished by switching "tools" or extruders.  The gcode for that varies a little depending on the flavor of gcode, but the most common reprap representation is the tool selector by itself on a line:

T0

or 

T1

And the color you get is whatever color filament is in that extruder.

alexrj, 2015-05-19:
There's `M160`, but I don't know any firmware able to read it, thus Slic3r doesn't generate it.

Since FFF printers generally work by alternating single-color filaments, they don't support RGB values. User assigns volumes/parts to extruders, and they are switched with `T0`, `T1` etc.
Feel free to ask if you have more questions.

closed 2015-05-19 by alexrj

columbus1231, 2015-05-20:
Thanks for your help. I will try your suggestion and see on my printer.