slic3r/Slic3r #2838: [Feature Request] Setting to automatically save and load a config bundle to keep different installations of Slicer in sync.

mikenew12, 2015-05-19:
I have Slic3r running on both a desktop and a laptop, and I constantly have to export and load the config bundle to keep them in sync. Would be great if there was a little script that ran on startup and on save that would save and load the config bundle from a particular location (dropbox, shared hard drive, etc.)

lordofhyphens, 2015-05-19:
what about a post-processing script so every time you sliced something?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2015-05-19:
@mikenew12, see the last paragraph of 
http://manual.slic3r.org/configuration-organization/configuration-organization

Maybe something more user-friendly can be worked out, but this works fine.

mikenew12, 2015-05-19:
@alexrj Ah sorry I didn't see that. I agree it might be nice to have that feature as part of the gui, but for now I'll just go ahead and symlink the settings files. Thanks for the response.

closed 2015-05-19 by mikenew12