slic3r/Slic3r #2841: Selectively disabling supports?

kefir-, 2015-05-22:
I thought this had been discussed before, but I couldn't find a reference when searching the issue tracker.

I've been trying to print a part that requires supports. However, there are some small cavities that are not easily accessible where the support is very difficult to remove. Using another slicer, I could enable support material but remove the supports in the affected areas, and the part came out fine. In slic3r, however, I can't find a way to achieve this.

I'd think it'd be possible to do this with modifiers. If slic3r had an option for this, it could for example skip building supports for overhangs inside the modifier. I think it should still build supports through the modifier if there's an overhang over the modifier, but if it could skip building supports for overhangs inside the modifier, I think we'd have a usable solution.

Mikk36, 2015-05-22:
This should be covered with this feature request: https://github.com/alexrj/Slic3r/issues/2570

lordofhyphens, 2015-05-22:
Modifier meshes allow you to do this in slic3r, it is in the 1.2.x branch.

You will need to generate a stl for the area you want. Anyone who wants to
make a better ui is welcome.

kefir-, 2015-05-22:
@Mikk36, thanks, I knew I'd seen this somewhere... 

@lordofhyphens Has this already been implemented? I use 1.2.7-dev, and my first thought was to use a modifier mesh, but I couldn't find the relevant setting for the modifier. Did I miss it?

lordofhyphens, 2015-05-22:
I could be derping hard there, unfortunately. I will take a look later.
Would be straightforward to implement though.

mikenew12, 2015-05-22:
I think you can only set support parameters at the object level, not with a modifier mesh. I would love to see this implemented as well since it would let us get around at least some of the problems with support generation.

lordofhyphens, 2016-11-27:
See my infill-over-plate branch/PR #3525 to partially cover this, those small cavities wouldn't be supported if there's no clear path straight down to the bed.

For the general case of modifier mesh + support parameters this is a dupe of #2570

closed 2016-11-27 by lordofhyphens