slic3r/Slic3r #2843: Skirt floating in the air in 3D preview

gege2b, 2015-05-24:
Hi all
Spotted a bug right now.
It happens when I set more than 1 raft layer (other config or model independant)

The 3D preview shows the skirt on the upper raft layer only
The Layer view is OK

The exported Gcode doesn't contain skirt at all

Debian Jessie 64b
Current dev

config : https://dl.dropboxusercontent.com/u/34934879/config-skirt.ini
example gcode : https://dl.dropboxusercontent.com/u/34934879/octopus_body.gcode

[image]

[image]

and on the next layer :

[image]

alexrj, 2015-05-24:
Hah, nice catch there...

alexrj, 2015-05-24:
[image]

alexrj, 2015-05-24:
Fixed. Thank you for the report.

closed 2015-05-24 by alexrj