slic3r/Slic3r #2846: "--use-firmware-retraction is not compatible with --wipe" when both options are set to no

andrewhodel, 2015-05-24:
It seems that the checking is done on the level of if the parameter for the option is passed and not actually on the value.

Can you fix that?

alexrj, 2015-05-24:
Can you provide more detailed steps?

andrewhodel, 2015-05-24:
command
```
~/Slic3r# ../localperl/bin/perl slic3r.pl --use-firmware-retraction no --wipe no
```

output
```
--use-firmware-retraction is not compatible with --wipe
```

It should not fail with this error, because the command is telling it specifically not to use either firmware retraction or wipe.

alexrj, 2015-05-24:
You're calling it the wrong way. Boolean options don't accept a value. You just turn them on with `--wipe` and off with `--no-wipe`.

closed 2015-05-24 by alexrj

andrewhodel, 2015-05-24:
What about this boolean that is default yes?

```
    --retract-layer-change
                        Enforce a retraction before each Z move (default: yes)
    --wipe              Wipe the nozzle while doing a retraction (default: no)
```

andrewhodel, 2015-05-24:
If so, according to what you said then --retract-layer-change would just be enabled what's already enabled?

alexrj, 2015-05-24:
As I said above: `--no-retract-layer-change`.

By the way, that brings to my attention that the --help page was not updated: retraction on layer change is now disabled by default.

andrewhodel, 2015-05-24:
If you could also update the README on the main GH page.

It would make sense to include the --no-* options in the --help output, right now there's no mention of it.

Although yet to me it seems strange to have an actual 2nd option instead of an argument passed to the same option for booleans.  All the other options have arguments.