slic3r/Slic3r #2847: 1.2.7: No way to close "Help -> About"

kefir-, 2015-05-24:
I updated Slic3r from git (revision 249a48d68d8bab5302fc8adbe4a304c3bffeb02a) and built it on Ubuntu 14.04. I clicked Help -> About, and then my screen looked like this:

[image]

I can't find a way to close the About screen, so I ended up having to kill Slic3r to get out of it. There's nothing to click, and I tried to hit ESC, Enter, and various other keys to remove it, but nothing seemed to help. Am I missing something, or is this a bug?

alexrj, 2015-05-24:
We'll probably need an "OK" button for handling that. See #2688 (I'm closing this one as you don't mind to avoid duplicates)

closed 2015-05-24 by alexrj

kefir-, 2015-05-24:
Sorry I missed that this was a duplicate. Thanks.