slic3r/Slic3r #2849: 1.2.6 missing perimeter segment on gcode export

hon-po, 2015-05-25:
On win 7 64 bit system, slice 20x20x5 cube, path on layers view look OK:
[image]
But exported gcode miss one perimeter segment:
[image]
This is repeating for all layers with fill density more than ~52%
Config:
https://drive.google.com/file/d/0B4bLJHgWDzxfdE5seGUxc1pKcDA/view?usp=sharing
stl:
https://drive.google.com/file/d/0B4bLJHgWDzxfN0JfaC1ENVVUdlE/view?usp=sharing
gcode:
https://drive.google.com/file/d/0B4bLJHgWDzxfMEtWVVRnak0wVVU/view?usp=sharing

lordofhyphens, 2015-05-25:
1.2.7 just came out, could you try that version and see if your problem
still exists?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2015-05-25:
Yes, there was a bug in an upstream library that causes that problem in 1.2.6.
This should be fixed as of 1.2.7. We'll wait for your confirmation, @hon-po. Thank you!

hon-po, 2015-05-25:
My bad, this is a repetier-host 1.0.6 not displaying the segment after G92 E gcode. I was trying pressure advance at 0.3. Craftware display the segment correctly.
[image]
No problem with both 1.2.6 & 1.2.7. Thanks!

closed 2015-05-25 by hon-po