slic3r/Slic3r #2850: 1.2.7: BadValue (integer parameter out of range for operation)

Findus23, 2015-05-25:
When starting Slic3r 1.2.7 in Debian Testing I get the following Error:

>The program 'slic3r' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 597 error_code 2 request_code 154 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

My system is the following.
```
lukas@debian:~/Slic3r1.2.7/bin$ uname -a
Linux debian 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt9-3 (2015-04-23) x86_64 GNU/Linux
```
Please contact me if you need more information. In the meantime I will try to install Slic3r from git.

Findus23, 2015-05-25:
Update:
When I compile from the latest git version (57e6a7becd9f101760750bd82804bdf5445e7cf8) everything works .

lordofhyphens, 2015-05-25:
I see you are using Debian; could you try to build a Debian package using
my branch https://github.com/lordofhyphens/Slic3r/ instead? I try to keep
it up to date with Slic3r HEAD and it uses Debian's libs instead of CPAN
(which I have seen to break stuff in weird ways).

alexrj, 2015-05-25:
@lordofhyphens, thank you for maintaining that fork. If you think that something could be merged upstream, let me know.

lordofhyphens, 2015-05-25:
No real thanks necessary, just scratching my own itch. It's got some warts
of its own atm because it's derived from the experimental package source;
hyperair packages stable releases and I grabbed his package source and
merged it with HEAD. As far as I know, the only changes in that fork are
the Debian-specific ones; anything that I think should go upstream gets its
own branch based off of HEAD.

Findus23, 2015-05-25:
My problem is not the build from Github, but the pre-built packages from dl.slic3r.org.

@lordofhyphens Discaimer: I'm not an expert in building packages and have no experience with perl.

I tired to build your fork. But the following error occurs:
```
lukas@debian:~/Slic3r_lordofhyphens$ LANG=C fakeroot debian/rules binary
: No such file or directory
'.  Stop. No rule to make target '
```

BTW: I also got an similar error when I try to run *slic3r.pl*:
```
lukas@debian:~/Slic3r$ LANG=C ./slic3r.pl 
: No such file or directory
```
But when I run
```
lukas@debian:~/Slic3r$ LANG=C perl slic3r.pl 
```
Slic3r starts without problems (so there seems to be something wrong with my perl)

lordofhyphens, 2015-05-25:
Why the explicit LANG reset? Just curious. Unless you have a particularly gnarly language setting, that shouldn't be necessary (mine is en_US.utf8). 

As for my fork, make sure you are in the debian branch; git might be trying to checkout the master branch (which doesn't have the debian dir in it). Also, you will need to have the package-building tools 
(see requirements from https://wiki.debian.org/IntroDebianPackaging) and fakeroot installed.

Findus23, 2015-05-25:
The LANG reset is just that the error message is displayed in English, in case you don't speak German.

The git repository was already on the debian branch (as it is the default branch) and I think I have all tools installed (I took the list from the README.md -> https://gist.github.com/Findus23/a787c7ee101a8194c86c)

I think the problem is that my computer is running perl-files with a different program (maybe bash) and therefore fails on a colon, when I execute the file via `./slic3r.pl`. When I run `perl slic3r.pl` it is obvious what program I want to use.

Findus23, 2015-05-25:
I found the mistake: The line endings are wrong (https://stackoverflow.com/questions/9651406/why-am-i-getting-no-such-file-or-directory-when-trying-to-execute-a-bash-scr)
I changed them (`find . -type f -exec dos2unix {} +`) and now the package is compiling

Findus23, 2015-05-25:
The package is completed. I installed the package and started slic3r. It is working except for the fact that no icons can be loaded and the following errors appear:
https://gist.github.com/Findus23/1b8fb3383e7772d636d8

lordofhyphens, 2015-05-25:
Ah, the shared bits didn't get installed... and slic3r is apparently using
/usr/bin again. Thought I had patched that.

It is patched now and pushed.

The package is completed. I installed the package and started slic3r. It is
working except for the fact that no icons can be loaded and the following
errors appear:
https://gist.github.com/Findus23/1b8fb3383e7772d636d8

jonaskuehling, 2015-05-26:
@Findus23 @alexrj Same error message here with 1.2.7-exp, running Ubuntu 15.04 64bit:

    The program 'slic3r' received an X Window System error.
    This probably reflects a bug in the program.
    The error was 'BadValue (integer parameter out of range for operation)'.
    (Details: serial 778 error_code 2 request_code 154 minor_code 24)
    (Note to programmers: normally, X errors are reported asynchronously;
    that is, you will receive the error a while after causing it.
    To debug your program, run it with the --sync command line
    option to change this behavior. You can then get a meaningful
    backtrace from your debugger if you break on the gdk_x_error() function.)

Findus23, 2015-05-26:
@lordofhyphens With your latest changes Icons are now displayed correctly.
Thanks for the help

simonkuehling, 2015-05-28:
@alexrj, just for the sake of completeness - i can confirm that 1.2.7-exp won't start as well with Ubuntu 14.04 64bit, same error message like @jonaskuehling 

Current git master compiled from source works just fine...

Is there any further system information i can provide for debugging?

alexrj, 2015-05-28:
Maybe try to remove some gtk-related .so lib in the binary package... I'm a bit clueless. Need to think about it :(

simonkuehling, 2015-05-28:
Hmm.. maybe i could try to package it as binary on my machine so we can check for differences?
Do you have the packaging process/script documented somewhere to give this a try?

alexrj, 2015-06-02:
I think this is the same thing as #2535. I think it's better not to ship any libGL.so, libGLU.sl or libglut.so with Slic3r and instruct users to install them using their distro's package manager. I'll do this when releasing 1.2.8.

Cyberwizzard, 2015-06-02:
I can confirm that on Linux Mint removing the libGL* files from 'dll' fixes the error and starts Slic3r 1.2.7.

alexrj, 2015-06-02:
Thank you @Cyberwizzard.

simonkuehling, 2015-06-02:
@Cyberwizzard @alexrj, confirmed - removing libGL* files from /dll fixes things here as well (Ubuntu 14.04)!

Findus23, 2015-06-09:
Quick update: Slic3r 1.2.8 is working for me.

alexrj, 2015-06-09:
Thank you!  We can close this.

closed 2015-06-09 by alexrj