slic3r/Slic3r #2851: 1.2.7 About Slic3r window cannot be closed.

hon-po, 2015-05-25:
Win 7 64 bit OS.
Missing button to close window, but focus remain on about window. Need to kill Slic3r to continue.
[image]

ignaworn, 2015-05-25:
I can confirm that also happens on Win10
Slic3r 1.2.7 experimental.

peter1960, 2015-05-25:
ALT-F4 will close it if your get stuck

hon-po, 2015-05-26:
Thanks!
ALT-F4 is good in closing the window.
Wonder should I close this issue?

a4jp-com, 2015-05-26:
I don't think closing this thread is a good idea yet. 

I think that the about screen should:

Have a close button somewhere,
it should close when you click on it anywhere besides the links,
it should close when you click any area besides the box,
or it should at least say press alt + f4 to close until it gets fixed.

alexrj, 2015-05-26:
There's #2688

hon-po, 2015-05-26:
I am closing this as it is already referenced in #2688 
I add (Windows OS) to the title, so it will be obvious in that thread. Thanks everyone.

closed 2015-05-26 by hon-po