slic3r/Slic3r #2852: 1.2.7 Crash when opening certain STL

esteevens, 2015-05-25:
When I add the following STL (https://www.dropbox.com/s/bsj5allwmhxu8sk/crash.stl?dl=0) slic3r crashes.  Opening it in 1.2.6 is working.

gege2b, 2015-05-26:
no crash for me (win7 x64 / 1.2.7 exp)
but without a more complete report[/url] this will be difficult to spot the problem
please read https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

gege2b, 2015-05-28:
ahah.
It crashes on my PC at home, but not on my PC at work :)
both are win7 x64

esteevens, 2015-05-29:
I tried to reproduce the issue again on my PC, but I cannot reproduce it anymore right now.  

Anyway here is my config
* Slic3r version: 1.2.7
* Windows 10 - Build 10074
* No specific command line parameters used.  Just launched Slic3r.exe.
* [Slic3r configuration file ](https://www.dropbox.com/s/a75mjv7ooocbu00/config.ini?dl=0)

Loading this [STL file](https://www.dropbox.com/s/bsj5allwmhxu8sk/crash.stl?dl=0) caused slic3r.exe to crash.

I'm not sure if the following details from the windows eventlog are usefull:
>Faulting application name: slic3r.exe, version: 1.2.7.37, time stamp: 0x503fe867
>Faulting module name: perl514.dll, version: 0.0.0.0, time stamp: 0x4f863eff
>Exception code: 0xc0000005
>Fault offset: 0x000000000003e61d
>Faulting process id: 0x2ed4
>Faulting application start time: 0x01d0970357aac574
>Faulting application path: C:\data\es\3D printer\Slic3r\slic3r.exe
>Faulting module path: C:\data\es\3D printer\Slic3r\cpfworkrt\perl514.dll
>Report Id: 20e7c1d6-d41e-4e1e-b5c9-1823de42194b
>Faulting package full name: 
>Faulting package-relative application ID:

closed 2023-07-11 by esteevens