slic3r/Slic3r #2856: Request: offset between lines in solid infill to improve layer adhesion

rsatue, 2015-05-26:
Hi,

I am working on a part in which it is necessary to have some solid infill because it is mechanically solicited. 

Currently slic3r is printing each layer over the previous one resulting in the following: 
[image]

It would probably improve the mechanical response of the part if we could print the layer with an offset equal to half the extrusion width. The result would look this way:
[image]

It would be necessary to increase a little the amount of material extruded but I think it could result in an improved print. 

What do you think?

Of course this option would only apply to solid infill layers. 

Thanks in advance.

closed 2015-05-26 by rsatue

alexrj, 2015-05-26:
Actually, solid infill layers are cross-hatched (i.e. each layer is rotated by 90°), so they don't reflect your first drawing.

whosawhatsis, 2015-05-27:
What would be even better would be to print alternating lines at half a layer heigh offset, so that they are packed hexagonally rather than rectangularly. You also wouldn't get the problem with the half-extrusion-width gaps at the edges shown here.