slic3r/Slic3r #2858: Missing Filament color config causes crash on startup

pzl, 2015-05-26:
Slic3r version: aa8b6af (bug introduced in this commit -- previous commits work fine).
wxWidgets version wxWidgets 3.0.2, Wx version 0.9927

Platform: Arch Linux

### Steps to reproduce:

**Method 1**:

1. Use a .Slic3r directory, and slic3r.ini from a previous installation of Slic3r before this commit
1. Make sure 'expert' mode is enabled in slic3r.ini
1. install version aa8b6af or 310212e
1. Run `slic3r.pl`

**Method 2**:
(Backup your existing .Slic3r data directory somewhere else)

1.  Completely remove your .Slic3r directory
1. install version aa8b6af or 310212e
1. Run `slic3r.pl`
    - This should run successfully, and ask setup questions about your machine. Defaults are fine, or click cancel
1. Enable Expert Mode, and exit.
1. Run `slic3r.pl` again

**Results**:

Program immediately stops/crashes (no GUI ever shows), and gives the output:
```Can't call method "config" on an undefined value at /usr/share/perl5/vendor_perl/Slic3r/GUI/Tab.pm line 407.```

From what I can gather, at some point during filament config loading, the second parameter given to `get_preset_config`, that is, `preset` is null or empty, and the program crashes. 

The resulting `slic3r.ini` from the second method is pasted below, which should trigger the behavior:

```
# generated by Slic3r 1.2.8-dev on Tue May 26 13:19:04 2015
autocenter = 1
background_processing = 1
main_frame_maximized = 
main_frame_pos = 6,26
main_frame_size = 1341,509
mode = expert
version = 1.2.8-dev
version_check = 1

[presets]
filament = 
print = 
printer =
```

lordofhyphens, 2015-05-27:
And yes, it's commit aa8b6afe8ba02a0450d755bb49b15f98c53b4eb1 that causes this behavior. Reverting the commit clears the symptoms.

lordofhyphens, 2015-05-27:
I think what needs to happen is we need a clean way to assume some kind of default (or leave it blank and notify the user). 
I had a small patch that simply returned undef in MainFrame and Tab.pm , but that causes other problems, as everything would need to be able to understand if it got handed an undefined value.  I was able to get the UI to load and saved a color to the filament. I haven't seen any other new options that would be causing a crash. 

And my slic3r.ini has the presets configured.

alexrj, 2015-05-27:
I think this is not actually related to the color option. Slic3r handles new options gracefully (we add them all the time). This is probably a Linux-specific issue related to #2613. I can't reproduce it so far :(

lordofhyphens, 2015-05-27:
It's possible, although reverting the commit that added color filaments also unbroke things. So something related to that commit seems to be the culprit. My experimentation indicates that an undefined variable ($preset) is being passed to the get_preset_config subroutine. I don't know from where.

mikenew12, 2015-05-28:
I got a close, but not quite the same error:
`Can't call method "filament_colour" on an undefined value at /Users/mikenew/Projects/third-party/Slic3r/lib/Slic3r/GUI/Plater.pm line 449.`

I noticed that under the filament directory in settings I had a file called 
`PLA (white) (Michael’s MacBook Pro's conflicted copy 2015-05-27).ini` (I have my settings folder symlinked to Dropbox). Shortening the name to `PLA (white) (conflicted).ini` fixed it. I assume it's the `'` in the filename that caused the problem.

alexrj, 2015-05-28:
Yes @mikenew12, I think it's the ’ and this is not actually related to the Linux issue.

commit 23119e36 (2015-06-01): Fixes #2858
commit 5b82f093 (2015-06-01): Merge pull request #2873 from darenschwenke/master

Fixes #2858
closed 2015-06-01 by alexrj