slic3r/Slic3r #2859: Why is first layer so dense when first_layer_extrusion_width 100%?

Cylindric, 2015-05-27:
I have a simple 20mm cube, that I sliced with all the default profile settings in Slic3r 1.2.7.

If I leave the first_layer_extrusion_width at the default 200%, things look pretty normal for the first and second layers:

[image]
[image]

But if I set first_layer_extrusion_width to 100%, the first layer looks like this:

[image]

Is that normal?

I only really noticed because I was getting massive over-extrusion on my first layer, so thought I'd take off the default 200% first layer setting.

The reason it's a problem for me, is because when I use a large nozzle (tried with 0.8mm), I get what looks like a related problem.
Here's what layers 1 and 2 look like with all the default settings, apart from the nozzle changed to 0.8mm:

[image]
[image]

This results in massive over-extrusion on layer 1.

xoan, 2015-05-27:
When extrusion width is configured by a %, it is relative to layer height, so ¿what is your first layer height?

alexrj, 2015-05-31:
The behavior you're describing is correct and expected. As pointed out by the previous comment, when extrusion width is expressed in % it's relative to later height.
If you get overstuffing on first layer you need to raise your Z end stop a bit.

closed 2015-05-31 by alexrj