slic3r/Slic3r #2860: OSX: Add "cmd+,"-hotkey to open preferences

luclu, 2015-05-27:
for standard osx user-experience.

alexrj, 2015-05-31:
Thank you, I did it.

Yes, Ctrl and Cmd are bound by the GUI toolkit we use, but I don't know whether there's a standard shortcut for Windows so let's see if someone complains for this.

closed 2015-05-31 by alexrj

luclu, 2015-05-31:
:heart: cool! :)

mikenew12, 2015-06-01:
Are you sure this is right? The only preferences hotkey I've ever seen has always been '⌘,' (command comma). I just tested on the half dozen programs I have open right now, and that's they hotkey for each one.

luclu, 2015-06-01:
@mikenew12 yeah, you are totally right!
the GUI-toolkit in use makes ```Ctrl+,``` to ```Cmd+,```on osx and ```Ctrl+,```on win.

alexrj, 2015-06-01:
Exactly @mikenew12, that's what Slic3r now has.

mikenew12, 2015-06-01:
Ha, sorry guys. I completely misread the title here. Sneaky comma.