slic3r/Slic3r #2861: [1.2.7-exp] wheel zoom inverted in Layer tab

gege2b, 2015-05-28:
Hi all

Another little anoying thing spotted right now

The wheel zoom is inverted (and so, less intuitive) in layers tab, compared to 3D, and 3D Preview ones

Win7 x64, Slic3r 1.2.7 exp

cheers

alexrj, 2015-05-28:
Yes... also @simonkuehling noticed that.  The current scrolling direction is the same as other 2D programs like Adobe Illustrator. I guess that opposite conventions are used for 2D programs vs 3D programs. Not sure what are the recommended ones. Maybe it's just better to have them aligned whatever they are. Dunno.

simonkuehling, 2015-05-28:
i couldn't really find a universal convention in scroll wheel zooming direction either to date - but i agree that aligned behavior is probably the least obtrusive...

For my very own personal taste i would initially expect to zoom in when pulling the wheel into my direction - but that's not even consistent through 3D programs i use (actually 1:1 right now when switching through my currently open windows)..

gege2b, 2015-05-28:
I find the zoom in 3D / Preview tabs are more intuitive, but it's only my opinion ^_^

maybe add an option in slic3rs preferences ?

a4jp-com, 2015-05-28:
+1 for an option for people to choose the way the mouse wheel zooms.

Lenbok, 2015-06-20:
Ewww, the current direction is horrid! :-)

Web browsers zooming (ctrl-wheel) use zoom in on wheel push, same as slic3r 3D panes.

a4jp-com, 2015-06-20:
I think the only problem is the "3D" and "Preview" tab zoom one way, but the "Layers" tab is reversed. If they are all zooming the same way it would be nice. Then having an option to reverse all or each direction in a settings area would make Slic3r even better, being slightly customizable. 

You can now press ctrl and zoom with the mouse wheel but I definitely don't want to have to push ctrl to zoom every time. Having it the way it is now is nice though to support everyone. The mouse wheel works well the way it is now. Zoom with the mouse wheel then pan by pressing down the middle scroll wheel.

ghuser2, 2015-09-22:
My 2D tab does not zoom at all. Nobody mentioned it, so that's probably how it is.

Lenbok, 2017-03-11:
Addressed by PR #3749

lordofhyphens, 2017-03-11:
@ghuser2 FYI, it's the 2D layer preview tab that has a zoom function.

commit 25bcff36 (2017-03-11): Fix #2861, provide consistent mouse-wheel zoom direction, plus a GUI preference to invert (#3749)

* Make mouse-wheel in 2D toolpath zoom in the same direction as it does in 3D and 3D-preview

* Add a GUI preference setting for whether to invert the direction that
mouse-wheel scrolling will zoom in the 2D and 3D panels.
closed 2017-03-11 by lordofhyphens