slic3r/Slic3r #2862: Retraction is miscalculated in g-code

klauzes, 2015-05-28:
Hello,
It seems that under printer settings / Extruder 1 / Retraciton :
Length, though specified in mm, is actually interpreted in g-code as milimeters/10 ? ...roughly
Extra Lenght on restart: is the sum of Lenght + Extra Lenght on restart 

As a test, I've set the Lenght =10 and Extra_lenght_on_restart= 2 and got the following g-code:

G1 X81.829 Y86.550 E14.57871 ; skirt
G1 X85.769 Y87.239 F720.000 E5.07871 ; 1800
G1 F1800.000 E4.57871 ; retract
G92 E0 ; reset extrusion distance
G1 X89.248 Y99.474 F900.000 ; move to first perimeter point
G1 E12.00000 F1800.000 ; compensate retraction
G1 X112.279 Y99.474 E12.52132 F420.000 ; perimeter

klauzes, 2015-05-28:
[image]

closed 2015-05-28 by klauzes

alexrj, 2015-05-31:
Exactly: 10 + 2 = 12 :-)

klauzes, 2015-05-31:
I thought I've closed this down.
The retract length is split between the wipe and retract, THAT threw me off. And adding something on top after retraction made no sense to me, but that doesn't mean it's useless. 
I'm writing my own gcode parse and have completely redesigned the electronics ... all < $20
High time to bring 3D printing to the masses, won't you say ?  :)
After I've doted my I's and crossed my T's, I'll release the first beta.
.Sorry to have wasted your time, mate

lordofhyphens, 2015-05-31:
We still see the issues if they are closed. :)