slic3r/Slic3r #2863: 1.2.7-Printer Setup TAB Change extruders from 1 to 2

billkaroly, 2015-05-29:
When I change Extruders from 1 to 2 it almost always crashes. I an using Windows 7 64 bit.

peter1960, 2015-05-29:
I just tried and see the same crash

billkaroly, 2015-05-29:
It happened in the previous experimental as well.

*Bill Karoly*
Product Manager
BoxBOT
*Email*: bill@boxbot.net
*Tel*: +1.801.784.8090

alexrj, 2015-06-01:
I'm not able to make it crash, however I noticed some concurrency issues that I just fixed. Is any of you able to test from git?

a4jp-com, 2015-06-09:
I have just updated 1.2.8-dev and went to the print settings area and changed the extruders from 1 to 2 and nothing crashed. All looks good here.

Windows 7 64-bit

alexrj, 2015-06-09:
@billkaroly, @peter1960, can you try with 1.2.8 and confirm this was fixed? Thank you!

billkaroly, 2015-06-09:
I'll give it a try. I'm using Windows 10 now though if that makes a
difference.  I can try it on another computer with windows 7 64 though

billkaroly, 2015-06-09:
Looks good in win10 64. So far.

closed 2015-06-20 by alexrj