slic3r/Slic3r #2864: 1.2.7 - Print Quality

billkaroly, 2015-05-29:
I am printing a small object at .2 layers and the top layer of the print in 1.2.7 looks nasty. Messy. Not smooth and defined as in 1.2.6. I am using the same settings.

In 1.2.6 the diameter of the printed part is 14.9 mm.
in 1.2.7 the diameter is 15.24 mm but the inside diameter is smaller.

The part I printed in the previous experimental I can slip over an 8 mm rod without diffiulty. The 1.2.7 printed part I have to do a lot of sanding.

The part is a tube with an outer dimension of 15.1 and inner ID of 8.2 in the stl file.

harriv, 2015-05-29:
Please post your config and model, as described in https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

alexrj, 2015-05-31:
@billkaroly, if you have dimension problems see:  
http://manual.slic3r.org/troubleshooting/dimension-errors

From what you wrote, there's no evidence of any bug whatsoever, but just the need for better tuning of your configuration (different Slic3r versions might require different settings). People are getting beautiful prints with 1.2.7.

If you want other users to have a look at your test case, you need to post the STL file and your full settings and any other detail that might help people help you.

closed 2015-05-31 by alexrj