slic3r/Slic3r #2865: Feature request : more descriptive gcode comments

zeograd, 2015-05-30:
Hi,
  in order to better integrate postprocessing scripts (like one I'm currently working on), I'd like to see more information leaked via the verbose gcode mode.

  As of now, all perimeters are attached the comment ' ; perimeter' without details whether they are the most external shell (or not) nor whether they are shells for the outside of the shape or shells for the inside (for the holes in the model).

  As I'd like to distinguish those different perimeter types, I've added the following trivial patch for figuring out which perimeter are external and not, and I assume that the first external perimeter from a layer is the outer one, non external perimeters up to the next external one are extra outer perimeters. But I think it would be more robust to generate gcode comments telling explicitely in which situation we are.

  https://gist.github.com/zeograd/fd4fd0d4e67ecb1a463f

I propose to have those 4 comments in gcode:
; perimeter external outer
; perimeter outer
; perimeter external inner
; perimeter inner

(or any equivalent rewording).

I can try to work on a patch for doing so if pointed at the right direction to distinguish outer from inner shells.
Thanks in advance.

lordofhyphens, 2016-07-27:
I don't know about differentiating that far down, but I believe that it would be very helpful to at least call out the printing moves that are already covered/tweakable in settings for speed, etc.

So that would be:
* external perimeters
* small perimeters
* other perimeters
* infill
* solid infill
* top solid infill
* support
    * skirt
    * brim
* support interface
* gap fill
* bridges
* travel
* Move to layer

Most of these are already present.

lordofhyphens, 2016-07-27:
Related: #3132