slic3r/Slic3r #2868: Crash when launching on Ubuntu 15.04 64 bits

avalero, 2015-05-31:
PC: DELL XPS 13
Ubuntu Version: 15.04 64 bits
Slic3r Version: 1.2.7

Error message:
The program 'slic3r' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 541 error_code 2 request_code 154 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

Stable version of Slic3r works properly

avalero, 2015-05-31:
When compiling from code it works

alexrj, 2015-05-31:
Hello, merging with #2850

closed 2015-05-31 by alexrj