slic3r/Slic3r #2870: 1.2.7 Exp: Ooze prevention with tall skirt and delta T stops retracting

kyrreaa, 2015-05-31:
Whenever I enable ooze prevention and set a delta temp I notice I get horrible ammount of oozing right outside the skirt. This becomes so heavy that it crashes the hotend in the end.
I looked in the generated g-code and there is no retract whenever it moves off to change temp.
Is there not supposed to be retract here, and if not, could we add this?
A checkbox or a value for retract in ooze prevention would probably be a good idea as it wouldn't necessarily be the same retract as when doing a quick move.

kyrreaa, 2015-06-01:
By the way, is there a way to set how far outside the skirt in any direction the head will wait?
My firmware does the nozzle alignment so I cannot tell Slic3r what my offsets are, then I get double offsets. Ideally I want both heads (all 3 really) to be outside the skirt while changing temps.

alexrj, 2015-06-01:
Actually, that's not oozing: that's active extrusion performed for purging the nozzle. The strings are expected to be cut/wiped when crossing the skirt going back into the print area.

The second thing you report is an actual issue: ooze prevention is designed to bring extruders outside skirt. But that relies on the extruder offset values, which unfortunately are set to 0,0 when letting the firmware handle this. This needs a fix.

kyrreaa, 2015-06-01:
Interesting!
It also does z-lift which makes it not wipe effectively as it passes the skirt...

How quickly could we get a "assume x radius to clear skirt" kind of setting?
Would it work if I set BOTH extruders to for example 10mm offset same way? (aka does it calculate clearance based on max offset as a radius in a circle, or in some other way I can exploit as a workaround.)