slic3r/Slic3r #2871: 1.1.7 : AMF files use material id="0"

bobc, 2015-05-31:
The AMF file produced contains material spec for id ="0" : 

  <material id="0">
    <metadata type="Name"></metadata>
  </material>
  <material id="1">
    <metadata type="Name"></metadata>
  </material>
  <material id="2">
    <metadata type="Name"></metadata>
  </material>

According to AMF spec http://en.wikipedia.org/wiki/Additive_Manufacturing_File_Format#Material_specification material id = "0" is reserved for void spaces (i.e holes). The wiki article was a bit vague but the ASTM spec makes it clear "0" is reserved.

Therefore, material ids should be numbered from "1".

alexrj, 2015-06-01:
Thank you for pointing that out.

alexrj, 2015-06-01:
Starting from 1.2.x Slic3r does not use AMF materials anymore but it puts config options inside <object> and <volume> metadata tags.  Files with materials are still read for backwards compatibility. I added a comment in the source code that will serve as a reminder should we start using materials anytime in the future. Thank you for your report.

closed 2015-06-01 by alexrj