slic3r/Slic3r #2872: When "combine infill" > 1, solid top layers should use combined infill height rather than perimeter height

triffid, 2015-06-01:
Yesterday I did a print that has a very shallow dome. I set "combine infill" to 5 layers and layer height to 0.05, which should give infill height of 0.25. This is the perfect usage case for this wonderful feature.

Unfortunately, I didn't also change "solid top layers" to 20-25, and slic3r seemed to try to put 0.05mm fill on top of 20% honeycomb fill which just plain does not work - see #240 and [images of terrible top-layer acne](http://imgur.com/UZJBp3A,KEE8OLQ)

**To me it would make more sense for slic3r to use *infill* height for solid top fill (ie 0.25mm) in this situation, rather than 2-20 layers of 0.05 solid fill.** The retreating perimeter line should cover the coarse infill neatly in this case.

Perhaps, when slic3r detects that combined fill is intersecting other combined fill with a different offset (eg infill at 2.2 and infill at 2.25 due to domed top) it could lower the higher one, and put some thin (0.05) fill on top to bring it up to height?

This might help avoid having [little bits of infill scattered all over the place](http://imgur.com/Hq8lJkr).

Would detecting domes in the model by searching for large connected groups of facets with not-quite-vertical normals help?

[config](http://triffid-hunter.no-ip.info/abs_new.ini)/[STL](http://triffid-hunter.no-ip.info/fin_v008-fixed.stl)/[Gcode](http://triffid-hunter.no-ip.info/fin_v008-fixed_ABS_EW0.6_LH0.05_ND0.35_FD20%25.gcode)/ 310212ed30b593c86ef502849e8c9c93050a3f49

alexrj, 2015-06-01:
We used to do that, but another report pushed us to limit the effects of "combine infill" to the internal infill. So we're in a "perpetual regression" loop. :-)

We can make that configurable. Or smarter, somehow.

harriv, 2015-06-01:
What if top layers were millimeters instead of layer count?

triffid, 2015-06-01:
I would like quite a number of settings to accept millimeters, eg perimeter
width, top/bottom infill as you say, maybe convert "combine infill" to
"infill layer height (mm)"

I think having top/bottom fill as millimeters rather than layers would
help, but we'd (apparently) still need a setting for whether to use infill
layer height or perimeter layer height for filling this volume.

kefir-, 2015-06-01:
I'm personally not a fan of specifying these things in mm, I prefer layer count. If course, if slic3r accepts both, that's fine. But slic3r could offer to recalc perimeter settings and top/bottom layer counts to keep the same thickness when layer height or other settings are changed that affect them.

mrvn, 2016-06-23:
@kefir- I think neither thickness in mm not layer count is the right option, you need both (and more?).

- For structural reasons and color/transparency you need a certain thickness for the top/bottom fill. This is best expressed in mm as the total thickness.
- Again for structural reasons and quality you need a certain number of layers. E.g. doing just one thick layer won't give a stable top fill while 2 half thick layers with perpendicular directions will be much more stable.
- For top fill the first (few) layer shouldn't count since it will be a bridging layer and sag in between the infill strands. For the first top fill (bridging) layer the thickness can't be controlled since there is nothing below to squish the filament flat against. The bridging layer should not count towards top fill thickness / layer count.

So I would suggest 3 settings:

- minimum solid top thickness
- minimum solid top layers
- extra solid top bridging layers

That way the same settings could be used for any number of layer heights.

MortarArt, 2018-05-31:
I feel like with the adaptive quality settings in place, the request for measurement based top and bottom layer heights is even more important.