slic3r/Slic3r #2874: Slic3r doesn't start if expert mode selected

einon, 2015-06-01:
Running Debian Jessie (Linux  3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt9-3 (2015-04-23) x86_64 GNU/Linux)
To repeat the issue:

Checkout the current master HEAD before the 'experimental autospeed feature' that causes a segfault (6e280ab), build and start via ./slic3r.pl
Select menu option File -> Preferences, and change Mode to 'Expert.
Restart the program.

Program is expected to start in expert mode, but fails with the error:

Can't call method "config" on an undefined value at /source/Slic3r-git/lib/Slic3r/GUI/Tab.pm line 407.

I've bisected this to the recent commit aa8b6af  'Assign color to filaments' causing the issue.

alexrj, 2015-06-01:
Thank you @einon. Can you check now?  I think I have fixed this.

einon, 2015-06-02:
Yes, it now starts in expert mode. Many thanks!

closed 2015-06-02 by einon