slic3r/Slic3r #2875: Mac releases not code signed

magu, 2015-06-01:
Since Mac releases of Slic3r are not code signed with an Apple Developer ID, Mac users cannot run them without bypassing [Gatekeeper](https://developer.apple.com/developer-id/) by either disabling it or right clicking and selecting open manually. This is easily corrected by code signing all releases with a Developer ID, making it possible to run the Slic3r directly on most users computers, despite Gatekeeper being enabled.

Currently, checking code signage shows lack of signature
```bash
$ codesign -vvv -d Slic3r.app/
Slic3r.app: code object is not signed at all
```
but after resigning it with my personal Developer ID
```bash
$ codesign --force --verify --sign "Developer ID Application: Magnus Kvevlander" Slic3r.app/Contents/Frameworks/lib* Slic3r.app
Slic3r.app/Contents/Frameworks/libwx_baseu-3.0.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_baseu-3.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_baseu_net-3.0.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_baseu_net-3.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_adv-3.0.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_adv-3.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_core-3.0.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_core-3.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_gl-3.0.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_gl-3.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_html-3.0.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_html-3.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_media-3.0.0.dylib: replacing existing signature
Slic3r.app/Contents/Frameworks/libwx_osx_cocoau_media-3.0.dylib: replacing existing signature
$ codesign -vvv -d Slic3r.app/
Executable=/Applications/Slic3r.app/Contents/MacOS/slic3r
Identifier=Slic3r
Format=bundle with Mach-O universal (i386 ppc7400)
CodeDirectory v=20200 size=1250 flags=0x0(none) hashes=56+3 location=embedded
Hash type=sha1 size=20
CDHash=2ca2628f2a348a0dcc83c63be9cff9c5f2a3270f
Signature size=8519
Authority=Developer ID Application: Magnus Kvevlander
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=1 jun 2015 16:19:29
Info.plist entries=10
TeamIdentifier=64GXNP98QK
Sealed Resources version=2 rules=12 files=135
Internal requirements count=1 size=168
```
it runs directly without any Gatekeeper message.

magu, 2015-06-01:
Stock Slic3r shows this when trying to run it
[image]
and it cannot be run directly, whereas it after resigning shows like this
[image]
and you can easily just select Open.

I know this is not _entirely essential_, but it easily adds additional polish to the user experience for Mac users.

paulkaplan, 2015-06-05:
This would require an apple developer account which cost money. I wonder if they allow open source projects to get free accounts to sign apps.

magu, 2015-06-05:
I do have a vague recollection that free Developer IDs for open source was supposed to be the case. However, now that I google around I can't find anything about that.

What is odd though, is that my Mac Developer Account expired at least a year ago, but my Developer ID certificate is valid until 2017, so they do not seem to be 100% connected. Maybe Alessandro can drop Apple an email and ask for a certificate as an open source developer?

a4jp-com, 2015-06-05:
Apple gives free developer accounts to companies or people that can fix problems with popular apps or programs.

Just tell them the situation and they might be able to help you.

magu, 2015-06-05:
If they grump, I could also lend my credentials to the project. I rely so much on Slic3r that I could give something back to the community.

lordofhyphens, 2017-03-29:
Apple has been paid its tax. Signing should now commence.

alexrj, 2017-03-29:
Not yet working...

lordofhyphens, 2017-03-29:
Oh, sorry.

alexrj, 2017-03-29:
Done! Automated OSX builds are now codesigned, and available at http://dl.slic3r.org/dev/
Thank you for the input, @magu!

closed 2017-03-29 by alexrj

magu, 2017-03-29:
Thank you for fixing it!