slic3r/Slic3r #2877: Possible Slic3r error when printing with Pressure Advance

ML3D, 2015-06-01:
I was attempting to print a part with Slic3r 1.2.7 using a Pressure Advance setting of .5. 

It previewed fine in Slic3r, however, the preview was slightly different in Repetier v1.0.6.

I printed it anyway, but oddly, in the same area as the preview errors, I had an even worse print error. Basically, certain layers of filament came out extremely sparse, causing a serious weak point. I stopped the print. When I removed the .5 from the Pressure Advance field, the preview in Repetier then matched what I saw in Slic3r. I don't know if this is a Slic3r issue or not, but I wanted to mention it.

Interestingly, both errors align with the top edge of a horizontal cylinder shaped cutout.

Please see the attached images. In the photograph you'll see the two areas where there is a filament gap that runs around the perimeter of the object.

If you need additional information, I'll provide it.

[image]

[image]

[image]

[image]

alexrj, 2015-06-02:
Thank you.   The preview error in Repetier-Host is an issue of Repetier-Host (and any other G-code based preview tool which doesn't know about the advance algorithm).

The voids in the actual print just mean that you might want to try different or smaller values. It's not easy to calibrate, unfortunately.

As soon as I release 1.2.8 you might want to try the new autospeed feature which works better than pressure advance.

closed 2015-06-02 by alexrj

ML3D, 2015-06-02:
Ok, thanks for the information. Looking forward to testing your autospeed feature. As always, thanks for your dedicated and hard work.