slic3r/Slic3r #2878: Connecting skirt to brim

foulowl, 2015-06-03:
Expected: 
Setting skirt "distance to object: 0" with a brim causes the skirt to be attached to brim.
 
Actual:
Skirt is roughly 1 mm away from brim. 

Use case:
My skirt pops off my build plate when it reaches a certain height, due to uneven cooling. Attaching to the brim would solve this issue. (Attempting to "Print perimeter surrounding skirt full height of part to keep the heat in." as per: http://reprap.org/wiki/Print_Troubleshooting_Pictorial_Guide#Too_cold_2 )

Alternative:
Setting skirt to "-1 mm" or similar would be acceptible as well, as long as it is possible to atttach skirt to brim I am happy.

Thanks!!

linuxlurak, 2015-06-04:
Do you have something like this?
[image]

markwal, 2015-06-05:
But, isn't just adding another extrusion width to the brim and turning off the skirt the same thing?

foulowl, 2015-06-08:
I have something like this:
http://imgur.com/ZNStf5z
Notice how this is with skirt distance to object set to 0.

Please note that this is not adding an additional extrusion width to the brim, because the skirt is as tall as the object in this case, whereas the brim is a single layer only.

Thanks!

markwal, 2015-06-09:
Ah.  I see.

rkertesz, 2015-08-14:
Agreed. Seeing same phenomenon