slic3r/Slic3r #2879: Hovering Supports

MartinMajewski, 2015-06-03:
Slic3r generates supports in free space that don't support themself.

https://dl.dropboxusercontent.com/u/4928832/HoveringSupports.png
https://dl.dropboxusercontent.com/u/4928832/HoveringSupports2.png

Slic3r 1.2.8dev (Commit No. c5f72a6)

Zuborg, 2015-06-04:
...and often ignores its duty to generate support:
[image]

Config 1.2.7: https://drive.google.com/open?id=0BzAa6v5hnj3fa2VNOHhFRXpJWGc&authuser=0
STL: https://drive.google.com/open?id=0BzAa6v5hnj3fN3ZBZ2UxMmFZVzQ&authuser=0

mikenew12, 2015-06-04:
@Zuborg This looks like this issue: #2839

whosawhatsis, 2015-06-04:
@Zuborg The problem here is that there are two surfaces, neither of which exceeds the overhang threshold angle. The support algorithm isn't smart enough to realize that although there are not strictly any /surfaces/ that need to be supported, there is a /structure/ that does need support.

A better support algorithm would actually look at the cross-section for each layer, subtract the cross-section for the previous layer, look for any features left that are larger than, say, half an extrusion width, mark those as areas that need support, then apply some minimum support structure contact width logic so that the support structure won't end up being thin enough to break or contacting a small enough area of the print that it will fall off.