slic3r/Slic3r #2881: 1.2.8-dev slic3r crashes when opening Help -> About

kefir-, 2015-06-03:
I'm running on Ubuntu 14.04, and when I try to open Help -> About in latest 1.2.8-dev, slic3r crashes and disappears. Output on the console is:

    Error while autoloading 'Wx::wxCLOSE' at /home/ketil/git/Slic3r/lib/Slic3r/GUI/AboutDialog.pm line 69.

alexrj, 2015-06-03:
Weird.  What wxWidgets version are you using? Can you paste the output of `slic3r.pl --debug --gui` ?

kefir-, 2015-06-03:
Sure!

    $ git rev-parse HEAD
    52d3a047a09ab45d755c8224e06b303687597563
    $ ./slic3r.pl --debug --gui
    wxWidgets version wxWidgets 2.8.12, Wx version 0,9927
    Data directory: /home/ketil/.Slic3r
    
    (slic3r.pl:2482): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 17
    
    (slic3r.pl:2482): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 17
    
    (slic3r.pl:2482): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 17

koghe, 2015-06-03:
here too, ubuntu 14.04, same slic3r/Wx version

alexrj, 2015-06-03:
Hrm I figured, wxCLOSE is not available in wxWidgets 2.8.  Let's see what I can find...

kefir-, 2015-06-03:
I tried fooling around with it, and changed it to wxCLOSE_BOX (because that's the closest name I found when I ran "strings Wx.so"), and with wxCLOSE_BOX it didn't crash, but then I still can't close the About-window. :)

kefir-, 2015-06-03:
Changing to wxOK works fine for me:

    $ git diff
    diff --git a/lib/Slic3r/GUI/AboutDialog.pm b/lib/Slic3r/GUI/AboutDialog.pm
    index 6fc83b6..78f89ef 100644
    --- a/lib/Slic3r/GUI/AboutDialog.pm
    +++ b/lib/Slic3r/GUI/AboutDialog.pm
    @@ -66,7 +66,7 @@ sub new {
         $vsizer->Add($html, 1, wxEXPAND | wxALIGN_LEFT | wxRIGHT | wxBOTTOM, 20);
         EVT_HTML_LINK_CLICKED($self, $html, \&link_clicked);
         
    -    my $buttons = $self->CreateStdDialogButtonSizer(wxCLOSE);
    +    my $buttons = $self->CreateStdDialogButtonSizer(wxOK);
         $self->SetEscapeId(wxID_CLOSE);
         EVT_BUTTON($self, wxID_CLOSE, sub {
             $self->EndModal(wxID_CLOSE);

a4jp-com, 2015-06-16:
Windows 7 64-bit works perfectly here. It must be an Ubuntu based problem.

strahlex, 2015-06-23:
Crashes on OpenSUSE and Debian Wheezy too.

GHPS, 2015-07-01:
The bug is still present in 1.2.9 stable under Kubuntu 14.04 LTS.

```
Error while autoloading 'Wx::wxCLOSE' at /usr/lib/slic3r/lib/Slic3r/GUI/AboutDialog.pm line 69.
```

PR #3166 (2016-03-13): Fix Slic3r crash when opening About dialog
fixes #2881 by replacing wxCLOSE button by wxOK
closed 2016-03-13 by alexrj