slic3r/Slic3r #2884: Is there a way to change printing speed at certain model height?

hippich, 2015-06-04:
I found a way to change layer height, but I need to change printing speed. Any idea how to do it?

alexrj, 2015-06-04:
That's not possible - you can just put a `M220` command manually where you want the speed change.

What do you need it for? Can you share with us your complete usage case?

hippich, 2015-06-04:
I am printing this model - https://github.com/hippich/stl/blob/94614f50656e88ec524d88093aa7abadd4add3f3/tv_cable_outlet_plug/tv_cable_plug.stl

But I do it with HIPS filament, which apparently needs slowdown significantly when doing overhangs. This is not a problem for bridges, as there is setting for it, but when doing sparse infill, like 25-40%, it doesn't stick well. In case of model I am printing perimeters closer to top become essentially overhangs and that's where I need to slow it down. So far I just slowdown whole print, but this is not necessary on the first 3/4 of a model, so I do it with adjusting feed rate manually on marlin lcd display.. And I hoped to be able to do it automagically, so I could leave it printing overnight.

[image]

On a related note - i thought about implementing it as a plugin, but then, correct me if I am wrong, there is no way to provide input for plugin, only original slic3r variables are available via ENV, right?

DeepSOIC, 2015-10-10:
maybe modifier mesh can help?

ghuser2, 2015-10-12:
I think modifiers are only for all-heights; they cannot be used on a per-layer basis like the Layers tab can.

lordofhyphens, 2015-10-12:
You can define a modifier mesh to be pretty much whatever shape and
thickness you want. If you know your layer height is 0.3mm and it's 15mm
off the ground, you can make an STL with those coordinates (IIRC) and bam!
you have a modifier mesh for that layer.

An OpenSCAD snippet for just this sort of thing is

thickness=0.2; # thickness of the mesh, should be an even multiplier of the
layer height
z=15; # also should be a legal layer starting point ((z -
first_layer_height) % layer_height = 0)
translate([0,0,z])
  translate([0,0,thickness/2]) cube([200,200,thickness],center=true);

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2015-10-12:
http://pastebin.com/SaxDfHGN <-- same as above, just on pastebin for all
eternity.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

DeepSOIC, 2015-10-12:
modifier meshes are wonderful.

[image]

ghuser2, 2015-10-12:
You mean if the modifier stl is modeled up off the ground, then when imported to slic3r it will appear up off the ground? Because currently when I import it gets set on the ground.

ghuser2, 2015-10-12:
Nevermind, yes, if modeled up off the ground, it does get placed up off the ground.