slic3r/Slic3r #2886: Error with STL & Illegal division by zero on practically everything

BurninSun, 2015-06-05:
Getting the following (or another error) on almost everything I try to slice at the moment:

Something is syntactically very wrong with this ASCII STL!: No error
Illegal division by zero at D:/Slic3r-master/lib/Slic3r/GUI/3DScene.pm line 281.

Freshly cleaned and compiled, using settings that I've used in the past.

Slic3r settings: http://pastebin.com/fPcEQ5x2
STL file: http://pastebin.com/B2b3XFnA

BurninSun, 2015-06-08:
Tried recompiling multiple times, keep coming up with the same problem. But considering no one else seems to be reporting this kind of issue, I'm thinking this might be something with my own system. The problem of course is that I've been running various dev builds with no problems for a few months, not sure why it suddenly doesn't want to work.

Anyhow, if it helps, here is the full steps I've taken including various log files. If anything stands out, let me know.

Windows 7, SP1

Deleted .cpan from user folder
Deleted entire Slic3r folder
Deleted entire citrusperl folder including gcc files

Went to
http://sourceforge.net/projects/citrusperl/files/Perl%205.16.3/Windows/
Downloaded 32-bit version due to the note about lacking FreeGLUT in the 64-bit version
citrusperl-standard-51603-msw-x86-018.zip

Unzipped and run
D:\citrusperl-standard-51603-msw-x86-018\perl\bin\citrusutils.exe
Hit the Set GCC button, "download the binary MinGW..."
Installed back into the same folder
Log file:
2015-06-08 16:29:24 : Message : MinGW was extracted to selected directory.
2015-06-08 16:29:24 : Message : The MinGW GCC path was set to D:\citrusperl-standard-51603-msw-x86-018\mingw32
and closed citrus perl

Run
D:\citrusperl-standard-51603-msw-x86-018\perl\bin\citrusterm.bat
at the prompt, run
cpan JSON App::cpanminus
Logfile:
http://pastebin.com/5atZWwWH

Download slic3r from
https://github.com/alexrj/Slic3r/archive/master.zip
and unzip then run
perl Build.PL
Logfile:
http://pastebin.com/9D8zsCWA
run
perl Build.PL --gui
Logfile:
http://pastebin.com/Mq48kDTr
run
perl Slic3r.PL

Hit Add... select stl:
http://pastebin.com/B2b3XFnA
(with configuration):
http://pastebin.com/fPcEQ5x2
and slic3r closes with error:
Something is syntactically very wrong with this ASCII STL!: No error
Illegal division by zero at D:/Slic3r-master/lib/Slic3r/GUI/3DScene.pm line 281.

alexrj, 2015-06-09:
I can reproduce the "Something is syntactically very wrong with this ASCII STL!" error with that STL file.

alexrj, 2015-06-09:
Thank you. I fixed this regression. I made the STL parser more robust for some things, but introduced this bug.

closed 2015-06-09 by alexrj

BurninSun, 2015-06-09:
Geez, something as simple as a space in the name. Would have never guessed.

Thank you very much.