slic3r/Slic3r #2887: Feature Request - "Start GCode" to be first or a "Init GCode" added

peter1960, 2015-06-05:
As my firmware powers off I need to always send a M80 to power on before the print.
The "Start GCode" is not the first lines in the file and always follow the M190/M104 commands.
As the power is not on it will wait forever at the the bed to heat-up and a reset is needed to get the power on and then print.
If the Start GCode was at the very start (like Cura) it would be good or an "Init GCode" area to be the very first.
Unless there is a configuration item to do this.

Mikk36, 2015-06-05:
Include M190 and M104 respectively in your start code.
If the temperature-setting commands are missing from your custom start code, Slic3r includes them by itself at the very beginning.

mikenew12, 2015-06-05:
@Mikk36 I tried that, but wasn't able to get it to work. It would still hang there waiting for the bed to start heating.

I found that sending `M140 S0` at the very start would keep it from waiting for the bed to heat up. So:
```
M140 S0
M80
M140 S[first_layer_bed_temperature]
```

Mikk36, 2015-06-05:
Not sure what exactly you tried, but this is the gcode I get from the start code I posted:
```
; generated by Slic3r 1.2.8-dev on 2015-06-05 at 09:50:27

; external perimeters extrusion width = 0.80mm
; perimeters extrusion width = 0.80mm
; infill extrusion width = 0.80mm
; solid infill extrusion width = 0.80mm
; top infill extrusion width = 0.80mm

M107
M80;
G21 ; Metric values
M107 ; Start with the fan off
M140 S0 ; Heated bed temperature
G28 ; Move X/Y/Z to min endstops
M109 S70 ; Preheat hotend for fans to kick in
M109 S200 ; Hotend temperature, wait
```

peter1960, 2015-06-05:
@mikenew12 , that looks good, thanks.
This can be closed

Mikk36, 2015-06-05:
You can close it yourself.

peter1960, 2015-06-05:
Sorry, have not used this before and did not know that was an option, Done

closed 2015-06-05 by peter1960

mikenew12, 2015-06-05:
@Mikk36 I just tried it again, and you're right; the g-code looks fine. Either there used to be a bug or I missed something the first time. Either way @peter1960, you may not need that first `M140 S0`.

peter1960, 2015-06-05:
Yes seems ok without the M140 S0