slic3r/Slic3r #2888: Low graphics quality on Mac with Retina display

kevinguibert, 2015-06-05:
Hi,

I'm a great fan of slic3r.

Recently, i changed my old Mac to a MacBook Pro with a Retina display.
When i installed slicer 1.1.7, i noticed that the GUI was in low graphics quality.

After some search over the web, i found that the app "Retinizer" was able to correct that and allow slic3r to have a high graphic quality on retina display (with a simple drag and drop of slic3r.app in the main window of the program).

I'm not good in programming but it might be really interesting for the development of Slic3r to use this app.

Best Regards

alexrj, 2015-06-07:
Can you post a screenshot of how it looks, please?
I don't have access to a Retina display.

Anyway, the appearance of the GUI is tied to the framework we use (wxWidgets), so it's something that should be reported to them as it's not in Slic3r's codebase.

You might also want to try a recent version like 1.2.7 which is nicer and is also compiled with a more recent version of wxWidgets.

kevinguibert, 2015-06-08:
Hi Alex,

I'm really sorry by posting this here since you told me it comes from the framework, but i will be pleased if i can help you.

Here are 2 screenshots from 1.1.7 (number 1 : without Retinizer app, number 2 with). I will post you some screenshot from the 1.2.7 which is having the same "low graphics display" (number 3 without, number 4 with)

[image]
[image]
[image]
[image]

mikenew12, 2015-06-08:
I think this is partially resolved in the dev build already. Here's a screenshot of 1.2.7, and then the same shot in the dev build.
[image]
[image]
The text and UI elements render at high resolution, but the icons don't. Obviously the icons aren't going to be high res unless there's also a @2x version included with the app. No idea if wxWidgets supports that or not.

kevinguibert, 2015-06-08:
Indeed, it seems that the dev version of Slic3r resolved that "low graphics display" issue. 
And as you said, the better way to get high res. icons is to implement @2x images.

Seems that my issue wasn't really a big issue !

alexrj, 2015-06-08:
Ouch, maybe I'm blind but I can't see any difference within each pair of screenshots you posted above!

Regarding the improvement that @mikenew12 mentions between 1.2.7 and dev (which I still fail to see from screenshots!) I think it's not a true change in Slic3r but a different behavior when you run a version which is compiled on another system (1.2.7, which I compile on OS X 10.6) and one compiled on your own system using native libraries.

(@kevinguibert, @2x is a CSS/web concept and does not apply to wxWidgets or GUI programming in general)
Unfortunately I have no Maverick or Retina to test or build on.

kevinguibert, 2015-06-08:
well i didn't think straight on screenshots. 
The only way to see if it's high res. or not is having... a retina display...

2 solutions might be possible then : 
1/ Compiling Slic3r on Maverick to get high def GUI
2/ Add a note to users with a link to Retinizer, if they want/need a high def GUI on Mac w/ Retina display

a4jp-com, 2015-06-08:
Can scalable vector images be used? I use them on my websites and don't have any scaling problems. I'm not sure how practical this solution is, but having something like wxWebKit embedded in your application and then using something like WebKit's SVG support to do the rendering if possible.

http://www.webkit.org/projects/svg/index.html

wxWebKit 
04/13/2013: This project is no longer under active development. Those considering embedding a WebKit browser in their wx application should consider embedding Chromium instead.

mikenew12, 2015-06-08:
@alexrj 10.6 predates retina displays, so I'd imagine you're right; the difference is the build environment. Here's a couple blown-up screenshots so you can see the difference. 
[image]
[image]

You actually can see the difference on the previous screenshots if you zoom in, even without a retina screen (or HiDPI if you want avoid the marketing speak). Just look closely at the text.

Also, a quick test shows that @2x assets are handled properly by wxWidgets, at least when building on my 10.10.3 machine. The @2x is a web convention, but it's also what Apple uses on OS X and iOS. I quickly drew a new 'red x' icon at 32 x 32 rather than the 16 x 16 of the original and added it to the 'var' folder as 'cross@2x.png'. Building and running on my non-retina machine shows the original, while the retina machine shows the higher-res icon.
[image]
[image]

Obviously this isn't a critical issue, but @kevinguibert is right, it really makes Slic3r much, much prettier on a retina mac. I do some OS X development, and I would be more than happy to package releases built against the latest SDK if you'd like. It would at least make the text and UI elements look better, and if you're able to get those icons at double resolution that would take care of everything.

kevinguibert, 2015-06-08:
Well, @mikenew12 your cross is definitively a high definition icon on a HiDPI (i'm trying to fit in) display !
As you said, not a critical issue, but it looks really nice on a Macbook Retina display !

alexrj, 2015-06-09:
I compile on 10.6 because that ensures Slic3r is compatible with older Macs. I'll think about possible solutions. All of them include finding a new set of icons that has 32x32 versions :)

mikenew12, 2015-06-09:
You should be able to build on 10.10 and set your build target to 10.6. And yes, higher res icons would be great :)

dragosmiron, 2016-06-12:
up

dragosmiron, 2016-06-12:
i dont think anyone has a low res mac anymore.

alexrj, 2017-04-04:
This should be fixed as of now. Get a fresh automated build at http://dl.slic3r.org/dev/mac/

closed 2017-04-04 by alexrj

gabrielmulle, 2018-07-17:
Hey folks, I see this was closed as resolved in 2017. I just downloaded this software version 1.2.9 and the low resolution graphics is still a thing. Any ideas?

michaelnew, 2018-07-17:
@gabrielmulle ...download the current release, not an old one?