slic3r/Slic3r #2889: Strange Bridging Detection Problem version 1.2.6

Mandlbaur, 2015-06-05:
I have a strange issue with a design of mine where one arm is correctly detected as a bridge, but the other is not.   http://www.thingiverse.com/thing:865652

[image]
[image]

Mikk36, 2015-06-05:
I believe this has been fixed already, not sure if it made it into the 1.2.7 or will come with 1.2.8 release.
If you'd like, it's already available in the git version.

Mikk36, 2015-06-05:
Yup, works as you'd expect in latest git version:
[image]

xoan, 2015-06-05:
It is fixed in 1ab8efb (see #2636)

Mandlbaur, 2015-06-05:
Thanks for the feedback guys

closed 2015-06-05 by Mandlbaur