slic3r/Slic3r #2890: Very messy prints.

a4jp-com, 2015-06-05:
Is this a setting problem, software or other problem? My prints are really messy for some reason.

I'm using 1.2.8-dev (Windows 7 64-bit)

[image]
[image]
[image]
[image]
The config files are not showing or selectable.

http://a4jp.com/3d-files/bugs/2015-06-06th-01-40-am-for-github-config.ini
http://a4jp.com/3d-files/bugs/deodorant-cap.gcode
http://a4jp.com/3d-files/bugs/deodorant-cap.stl

xoan, 2015-06-05:
This is an ooze issue not related in any way to the slicer software
itslef, but only to your configuration (retraction length/speed too
small/low, temperature too high, etc...)

a4jp-com, 2015-06-05:
Thank you Xoan for the super fast reply. Any good ideas for changing the settings to get slightly better prints?

Also I can't select the config files on the platter screen so I think that is a bug. I could select files ages ago. I didn't notice when this changed though.