slic3r/Slic3r #2891: Slic3r 1.2.7 Fails On Fedora 22

gralco, 2015-06-05:
Running just the `slic3r` x86 build works fine, but `slic3r --gui` says this

```
Can't load '/home/ekuzmenko/Downloads/Slic3r-1.2.7/lib/std/auto/Wx/Wx.so' for module Wx: libgtk-x11-2.0.so.0: cannot open shared object file: No such file or directory at <content>/DynaLoader.pm line 157.
 at /home/ekuzmenko/Downloads/Slic3r-1.2.7/lib/std/Slic3r/GUI/AboutDialog.pm line 6
Compilation failed in require at /home/ekuzmenko/Downloads/Slic3r-1.2.7/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
BEGIN failed--compilation aborted at /home/ekuzmenko/Downloads/Slic3r-1.2.7/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
Compilation failed in require at /home/ekuzmenko/Downloads/Slic3r-1.2.7/lib/std/Slic3r/GUI.pm line 8.
BEGIN failed--compilation aborted at /home/ekuzmenko/Downloads/Slic3r-1.2.7/lib/std/Slic3r/GUI.pm line 8.
Compilation failed in require at (eval 70) line 1.
```

The Slic3r x86_64 build doesn't work at all, it outputs:
```
(slic3r:11318): Gdk-ERROR **: The program 'slic3r' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 992 error_code 2 request_code 153 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)
```

gralco, 2015-06-08:
I apologize this is the same issue as was experienced in https://github.com/alexrj/Slic3r/issues/2535#issuecomment-109763048

Remove the packaged "**libGL.so.1.2**" from the dll directory fixed the problem, just using my distro's provided library works.

closed 2015-06-08 by gralco

lordofhyphens, 2015-06-08:
> I apologize this is the same issue as was experienced in #2535 (comment)

Yeah, I think that packaging opengl libs for Linux causes more issues than
it solves. Maybe just writing an error message for opengl if they don't
have it?