slic3r/Slic3r #2892: mach3 PWM fan control

rakolju, 2015-06-06:
mach3 can control the spindle rotation speed g code and convert it into a pwm output.
This feature can be used to 3d printer's cooling fan speed control.
if slic3r can add g code spinde speed commands
example:

g Code: s10 = (spindle speed = 10% 3D printer cooling fanspeed)

g code: s50 = (spindle speed = 50% of the 3D printer cooling fanspeed)

g code: s100 = (spindle speed = 100% 3D printer cooling fanspeed)

mach3 spindle configuration tap max speed 100rpm spindle and pulley ration 1

or

g Code: s25 = (spindle speed = 10% 3D printer cooling fanspeed)

g code: s125 = (spindle speed = 50% of the 3D printer cooling fanspeed)

g code: s250 = (spindle speed = 100% 3D printer cooling fanspeed)

mach3 spindle configuration tap max speed 250rpm spindle and pulley ration 1

I would like someone to the programmer to increase
this feature slic3r program?

rakolju, 2015-06-06:
this is the easiest way to manage your 3D printer fan speed automatically Mach3 program.

lordofhyphens, 2021-03-21:
Implemented with `fan_percentage` (0-100 range instead of 0-255).

closed 2021-03-21 by lordofhyphens