slic3r/Slic3r #2894: Please add purge walls to the software

DavidXanatos, 2015-06-07:
Please add purge walls to the software,
like in the image here: http://www.filaroll.com/blogs/news/13765039-advanced-3d-printing-material-store

kefir-, 2015-06-07:
Have a look at issue #2866, would that solve your problem? The version using [active_extruder] only works on current dev, otherwise wait for the next release, 1.2.8.

DavidXanatos, 2015-06-07:
It would be better if it would create such towers/walls autoamtically like makerware does.

rsatue, 2015-06-09:
Hi,

Will the solution proposed in #2866 work when you are using more than one extruder in the same layer?

Wouldn't make more sense to add some gcode lines to print the wipe-tower before every layer change and every change of tool?

lordofhyphens, 2015-06-09:
@DavidXanatos, in order to have a prayer of getting a feature added; you need to make a case for it. @alexrj does not have infinite time, and unless you submit a patch yourself to add the feature, it's probably not going to get done. "Program X does this" is not a reason. If Program X does what you want and @alexrj isn't convinced it needs to be done... you should probably go use Program X instead.

So, you need to lay out what your current real problem is, with pics, current slic3r configs and a model that exhibits the behavior that is undesired (see the contributing guidelines). Then we go to "how do we go about solving it".

DavidXanatos, 2015-06-10:
OK... so...

When a extruder is not used for some time (while the other prints) I observed that without the motor pushing still some material drips from the nozzle.
I guess it is caused by a non uniform heat distribution inside the extruder and possible super heating of the material what leads to bubbling/thermal decomposition of the filament.
When the extruder is used again there is a void inside and when the motor starts to push at first nothing comes out (or not enough).

This is especially problematic for materials like PVA which have a very narrow temperature window for printing.

There are two problems from this:
1. prints with extruder 2 are contaminated with material from extruder 1 and vice versa
2. as described above there are gaps in the extruded material what is really bad for the integrity of the object.

Now when such purge walls are printed by an extruder always before starting to print the current layer, possible voids in the nozzle caused by dripping are pushed out and problem 2 is fully avoided.
And since no extruder is not used for a long time the dripping is minimized, as it better maintains the heat distribution of a working extruder.

I don't have pics, I don't collect messed up prints. Are these really needed? I think the explanation above is clear enough.

"you should probably go use Program X instead"
Program X is a shit for other reasons so thats not an option.

David X.

phizz166, 2015-06-13:
It is completely possible to build purge walls automatically using the layer change and toolchange g-code functions. It wouldn't make a lot of sense to have the software do it automatically; there are so many different extruder and printer geometries that to do it right, you'd need nearly as many variables as if you just wrote the g-code yourself. 

Look around online...I've seen multiple people (try the reprap forums) giving examples of their code to build wiping towers in a print.

DavidXanatos, 2015-07-06:
Is it really so simple to add wipe towers by hand?
I would immagine that the code would need to know where the model is?