slic3r/Slic3r #2896: Feature request: Fixed multi-extruder order / Making tool change minimizing optional

iamnotachoice, 2015-06-09:
Slic3r has a nice way of minimizing tool canges in multi extruder setups. It will always start a layer with the extruder that has been used at the end of the previous layer. While this is nice and saves us some tool changes, it also introduces a wavy pattern wherever perimeters are interfacing, which also covers perimeters supported by dissolvable filament. Thats why, I would like to ask, if this tool change minimization could be made optional in the **Multiple Extruders** section of the **Print Tab**, allowing layers to be printed in a fixed extruder order (1st extruder, 2nd extruder, and so on).

The following should explain, why making this feature optional makes sense:

Since the FDM printer prints individual traces, each trace is squeezed next to another trace, we're aiming for a result that could look like this cross-section of a dual color print in green and blue, while blue could also be a dissolvable material like HIPS:

[image]

But in the real world scenario, there are traces, most importantly perimeters (and interfacing perimeters in multi-extruder-prints), which are not squeezed against another trace at print time, therefore "over-squeezing" a little:

[image]

Since the extruder, that has been used last in the previous layer will be used first in the next layer, the "over-squeezing" will now happen in the other direction:

[image]

And in the finished object, the interface walls will have a wavy pattern due to this

[image]

While in some cases, this does not hurt so much, in other cases, this is an issue. Steep overhangs and the use of dissolvable support material worsen the effects described here, as well as printer setups where the extruder offset is not an integral multiple of the printers positioning resolution.

If we woud dissolve the blue material from the example, we end up with this sort of wavy pattern on the perimeters of the green material:

[image]

I've made those graphics to illustrate the matter, but observed the issue on different printers by now. It is rather hard to capture it on camera. Basically, prints with 0.1 mm layer height look to the eye as if they where printed with 0.2 mm layer height wherever the support material interfaces to the perimeters of the object. I solved the issue for me by using a Perl script that re-sorts the printing process to a fixed extruder order, but would appreciate if the feature, that was originally intended to minimize tool changes, would be optional.

phizz166, 2015-06-13:
I second this -- sometimes minimizing tool-switching is a good idea, and sometimes you want a simple sequential build. The checkbox is a good idea.

iamnotachoice, 2015-06-18:
Just published my script in it's own repo. Basically, it reorders the printing moves to a sequential build and adds wipe/prime towers. Instructions included:

https://github.com/iamnotachoice/Multi-Extrusion-post-processing-scripts-for-Slic3r/blob/master/README.md

Of course, I would also be glad to see the wipe/prime towers suggested by the script built-into Slic3r :)