slic3r/Slic3r #2897: The mouse over instructions disappear too quickly

a4jp-com, 2015-06-09:
I have version 1.2.8-dev on Windows 7 64-bit and the software is working quite well but the mouse over hints disappear too quickly so I have to keep moving the mouse over the buttons a few times to read everything. Can you make the hints stay longer?

a4jp-com, 2015-06-15:
There isn't enough time to read 4 lines of text.

[image]

a4jp-com, 2015-06-21:
Maybe double the time now would allow enough time to read the notes.

ghuser2, 2015-09-20:
This was also my biggest problem when first learning slic3r. I say it should stay until you move off of it.

a4jp-com, 2015-11-08:
Could the comments just stay open all the time?