slic3r/Slic3r #2899: Wrong infill in 1.2.8

rsatue, 2015-06-10:
Same part, same config.ini are resulting in different infills. Obviously the 1.2.8 infill is wrong.

Here is the part sliced with 1.2.7:
[image]

And here, with 1.2.8: 
[image]

When reading the gcode it confirms that in those points there are huge extrusions without any reason. 

The part file: https://www.dropbox.com/s/ptuz0lpd4fb0svj/bug%20part.stl?dl=0
The config file: https://www.dropbox.com/s/o1spfalo2arcqya/nozz06.ini?dl=0

alexrj, 2015-06-10:
Ouch.  Last minute change resulted in horrible effect.

alexrj, 2015-06-12:
Actually, this looks more like a rendering issue than a real infill issue as it's not noticeable in 2D toolpaths nor in other 3D preview tools. I'll investigate further.

alexrj, 2015-06-13:
Fixed. Thank you for reporting this.

closed 2015-06-13 by alexrj