slic3r/Slic3r #2901: [1.2.8] Small tower always stuck at 100% infill?

lordofhyphens, 2015-06-11:
Printing http://www.thingiverse.com/download:17276 with 0.4mm nozzle, 0.3mm layer height is giving 100% infill at every setting > 0%. Seems like everything is getting treated as rectilinear solid infill. 
0% infill works as expected.

Pic for reference.
[image]

Is this intended behavior, or did I trip over a regression involving infill (or somehow misconfigured things)? I don't remember seeing this behavior with earlier versions. 

Slic3r config:
 avoid_crossing_perimeters = 0
 bed_shape = 0x0,265x0,265x295,0x295
 bed_temperature = 100
 before_layer_gcode = 
 bridge_acceleration = 3000
 bridge_fan_speed = 100
 brim_width = 5
 complete_objects = 0
 cooling = 1
 default_acceleration = 9000
 disable_fan_first_layers = 1
 duplicate_distance = 6
 end_gcode = G10\nM104 S0\nG91\nG1 Z1\nG90\nG1 X0\nG28 Y0\nM84     ; disable mots\n
 extruder_clearance_height = 20
 extruder_clearance_radius = 20
 extruder_offset = 0x0
 extrusion_axis = E
 extrusion_multiplier = 1
 fan_always_on = 0
 fan_below_layer_time = 60
 filament_colour = #000000
 filament_diameter = 2.85
 first_layer_acceleration = 3000
 first_layer_bed_temperature = 100
 first_layer_extrusion_width = 1
 first_layer_speed = 30
 first_layer_temperature = 240
 gcode_arcs = 0
 gcode_comments = 1
 gcode_flavor = reprap
 infill_acceleration = 9000
 infill_first = 0
 layer_gcode = 
 avoid_crossing_perimeters = 0
 bed_shape = 0x0,265x0,265x295,0x295
 bed_temperature = 100
 before_layer_gcode = 
 bridge_acceleration = 3000
 bridge_fan_speed = 100
 brim_width = 5
 complete_objects = 0
 cooling = 1
 default_acceleration = 9000
 disable_fan_first_layers = 1
 duplicate_distance = 6
 end_gcode = G10\nM104 S0\nG91\nG1 Z1\nG90\nG1 X0\nG28 Y0\nM84     ; disable mots\n
 extruder_clearance_height = 20
 extruder_clearance_radius = 20
 extruder_offset = 0x0
 extrusion_axis = E
 extrusion_multiplier = 1
 fan_always_on = 0
 fan_below_layer_time = 60

 max_fan_speed = 100
 max_print_speed = 80
 max_volumetric_speed = 8
 min_fan_speed = 100
 min_print_speed = 10
 min_skirt_length = 0
 notes = 
 nozzle_diameter = 0.4
 only_retract_when_crossing_perimeters = 1
 ooze_prevention = 0
 output_filename_format = [input_filename_base].gcode
 perimeter_acceleration = 9000
 post_process = 
 pressure_advance = 0
 resolution = 0
 retract_before_travel = 2
 retract_layer_change = 1
 retract_length = 0
 retract_length_toolchange = 10
 retract_lift = 0
 retract_restart_extra = 0
 retract_restart_extra_toolchange = 0
 retract_speed = 30
 skirt_distance = 6
 skirt_height = 1
 skirts = 1
 slowdown_below_layer_time = 5
 spiral_vase = 0
 standby_temperature_delta = -5
 start_gcode = M106 S0\nM200 D[filament_diameter_0]\nG28 Y0 Z0 ; Home Z\nG90; set to absolute position\nG1 Z0.1; leave some room for oozing\nM104 S100; preheat nozzle.\nM190 S[first_layer_bed_temperature]; Wait until heatbed hits [first_layer_bed_temperature] \nM109 S[first_layer_temperature]; heat nozzle to [first_layer_temperature]C;\n; wait 2m?\nG1 Z5 F100 ; lift nozzle\nG11
 temperature = 226
 threads = 2
 toolchange_gcode = 
 travel_speed = 130
 use_firmware_retraction = 1
 use_relative_e_distances = 0
 use_volumetric_e = 1
 vibration_limit = 0
 wipe = 0
 z_offset = 0
 dont_support_bridges = 1
 extrusion_width = 0
 first_layer_height = 0.3
 infill_only_where_needed = 0
 interface_shells = 0
 layer_height = 0.3
 raft_layers = 0
 seam_position = aligned
 support_material = 0
 support_material_angle = 0
 support_material_contact_distance = 0.2
 support_material_enforce_layers = 0
 support_material_extruder = 1
 support_material_extrusion_width = 0.8
 support_material_interface_extruder = 1
 support_material_interface_layers = 3
 support_material_interface_spacing = 0
 support_material_interface_speed = 0
 support_material_pattern = honeycomb
 support_material_spacing = 5
 support_material_speed = 60
 support_material_threshold = 0
 xy_size_compensation = 0
 bottom_solid_layers = 3
 bridge_flow_ratio = 0.95
 bridge_speed = 60
 external_fill_pattern = concentric
 external_perimeter_extrusion_width = 0
 external_perimeter_speed = 50%
 external_perimeters_first = 0
 extra_perimeters = 1
 fill_angle = 45
 fill_density = 1%
 fill_pattern = honeycomb
 gap_fill_speed = 20
 infill_every_layers = 1
 infill_extruder = 1
 infill_extrusion_width = 0
 infill_overlap = 25%
 infill_speed = 80
 overhangs = 1
 perimeter_extruder = 1
 perimeter_extrusion_width = 0
 perimeter_speed = 60
 perimeters = 3
 small_perimeter_speed = 20
 solid_infill_below_area = 70
 solid_infill_every_layers = 0
 solid_infill_extruder = 1
 solid_infill_extrusion_width = 0
 solid_infill_speed = 20
 thin_walls = 1
 top_infill_extrusion_width = 0
 top_solid_infill_speed = 25
 top_solid_layers = 3

lordofhyphens, 2015-06-11:
If I'm just derping here, just close the issue.

kefir-, 2015-06-11:
Isn't this because of this setting? 

solid_infill_below_area = 70

Small areas are filled in according to this setting.

lordofhyphens, 2015-06-11:
Yep, that's what it was. Derp a herpderp.

closed 2015-06-11 by lordofhyphens