slic3r/Slic3r #2902: Compile error in XS.c: "expected ‘)’ before numeric constant"

giantallium, 2015-06-11:
> perl Build.PL --sudo
. . .
cc -I/usr/lib/perl/5.14/CORE -DXS_VERSION="0.01" -DVERSION="0.01" -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -fno-inline-small-functions -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o buildtmp/XS.o buildtmp/XS.c
buildtmp/XS.c: In function ‘void boot_Slic3r__XS(PerlInterpreter*, CV*)’:
buildtmp/XS.c:35553:5: error: expected ‘)’ before numeric constant
buildtmp/XS.c:35553:25: error: expected ‘)’ before ‘;’ token

slicer version 1.2.8

git rev-parse HEAD
4a91ea817aa15bafdb4cb2dac723361a95d9d072

BeagleBoard.org Debian Image 2015-03-01

Debian Version 7.8

lordofhyphens, 2015-06-11:
Try using my fork and building a package.

giantallium, 2015-06-11:
Thanks Joseph!  I will try that and let you know how it goes.

alexrj, 2017-03-18:
We can close this now.

closed 2017-03-18 by alexrj