slic3r/Slic3r #2904: 1.2.8: Failed test 'intersection_pl - result has same length as subject polyline'

arekm, 2015-06-11:
Test suite fails here on Linux x86_64, perl 5.20.1 with:

````
ok 18 - intersection_pl - result is not empty
ok 19 - intersection_ppl - result is not empty
ok 20 - intersection_pl - result is not empty
not ok 21 - intersection_pl - result is not empty

#   Failed test 'intersection_pl - result is not empty'
#   at t/11_clipper.t line 193.
#          got: '6'
#     expected: '5'
ok 22 - intersection_pl - result is not empty
not ok 23 - intersection_pl - result has same length as subject polyline

#   Failed test 'intersection_pl - result has same length as subject polyline'
#   at t/11_clipper.t line 206.
#          got: '92066651.9339821'
#     expected: '79586257.0728936'
# Looks like you failed 2 tests of 23.
Dubious, test returned 2 (wstat 512, 0x200)
Failed 2/23 subtests 
t/12_extrusionpathcollection.t .. 
1..18
ok 1 - 'collection object with items in constructor' isa 'Slic3r::ExtrusionPath::Collection'
ok 2 - no_sort is false by default
ok 3 - append ExtrusionPath::Collection
ok 4 - append ExtrusionPath
ok 5 - append ExtrusionLoop
ok 6 - 'correct object returned for collection' isa 'Slic3r::ExtrusionPath::Collection::Ref'
ok 7 - 'correct object returned for path' isa 'Slic3r::ExtrusionPath::Ref'
ok 8 - 'correct object returned for loop' isa 'Slic3r::ExtrusionLoop::Ref'
ok 9 - correct object returned for cloned path
ok 10 - correct object returned for cloned loop
ok 11 - appended collection was duplicated
ok 12 - items are returned by reference
ok 13 - chained_path_from
ok 14 - chained_path
ok 15 - chained_path_from
ok 16 - chained_path with no_sort
ok 17 - expected no_sort value
ok 18 - no_sort is kept after clone
ok
t/13_polylinecollection.t ....... 
1..3
ok 1 - chained_path_from
ok 2 - chained_path
ok 3 - chained_path_from
ok
t/14_geometry.t ................. 
1..9
ok 1 - 'convex_hull returns a Polygon' isa 'Slic3r::Polygon'
ok 2 - convex_hull returns the correct number of points
ok 3 - directions_parallel_within
ok 4 - directions_parallel_within
ok 5 - directions_parallel_within
ok 6 - directions_parallel_within
ok 7 - directions_parallel_within
ok 8 - directions_parallel_within
ok 9 - arrange() returns expected number of positions
ok
t/15_config.t ................... 
1..110
ok 1 - set/get float
ok 2 - serialize float
ok 3 - set/get int
ok 4 - serialize int
ok 5 - set/get string
ok 6 - serialize string
ok 7 - set/get string with newline
ok 8 - serialize string with newline
ok 9 - deserialize string with newline
ok 10 - set/get absolute floatOrPercent
ok 11 - serialize absolute floatOrPercent
ok 12 - set/get relative floatOrPercent
ok 13 - serialize relative floatOrPercent
ok 14 - set/get bool
ok 15 - serialize bool
ok 16 - set/get enum
ok 17 - serialize enum
ok 18 - deserialize enum (gcode_flavor)
ok 19 - deserialize enum (gcode_flavor)
ok 20 - deserialize enum (fill_pattern)
ok 21 - deserialize enum (support_material_pattern)
ok 22 - set/get points
ok 23 - set/get points
ok 24 - serialize points
ok 25 - deserialize points
ok 26 - deserialize points
ok 27 - reject undef points
ok 28 - set/get floats
ok 29 - serialize floats
ok 30 - deserialize floats
ok 31 - deserialize a single float
ok 32 - reject undef floats
ok 33 - set/get ints
ok 34 - serialize ints
ok 35 - deserialize ints
ok 36 - reject undef ints
ok 37 - set/get bools
ok 38 - get_at bools
ok 39 - get_at bools
ok 40 - get_at bools
ok 41 - serialize bools
ok 42 - deserialize bools
ok 43 - deserialize bools from empty string
ok 44 - deserialize bools from non-string value
ok 45 - reject undef bools
ok 46 - set/get strings
ok 47 - serialize strings
ok 48 - deserialize strings
ok 49 - reject undef strings
ok 50 - get_keys and as_hash
ok 51 - set/get float
ok 52 - serialize float
ok 53 - set/get int
ok 54 - serialize int
ok 55 - set/get string
ok 56 - serialize string
ok 57 - set/get string with newline
ok 58 - serialize string with newline
ok 59 - deserialize string with newline
ok 60 - set/get absolute floatOrPercent
ok 61 - serialize absolute floatOrPercent
ok 62 - set/get relative floatOrPercent
ok 63 - serialize relative floatOrPercent
ok 64 - set/get bool
ok 65 - serialize bool
ok 66 - set/get enum
ok 67 - serialize enum
ok 68 - deserialize enum (gcode_flavor)
ok 69 - deserialize enum (gcode_flavor)
ok 70 - deserialize enum (fill_pattern)
ok 71 - deserialize enum (support_material_pattern)
ok 72 - set/get points
ok 73 - set/get points
ok 74 - serialize points
ok 75 - deserialize points
ok 76 - deserialize points
ok 77 - reject undef points
ok 78 - set/get floats
ok 79 - serialize floats
ok 80 - deserialize floats
ok 81 - deserialize a single float
ok 82 - reject undef floats
ok 83 - set/get ints
ok 84 - serialize ints
ok 85 - deserialize ints
ok 86 - reject undef ints
ok 87 - set/get bools
ok 88 - get_at bools
ok 89 - get_at bools
ok 90 - get_at bools
ok 91 - serialize bools
ok 92 - deserialize bools
ok 93 - deserialize bools from empty string
ok 94 - deserialize bools from non-string value
ok 95 - reject undef bools
ok 96 - set/get strings
ok 97 - serialize strings
ok 98 - deserialize strings
ok 99 - reject undef strings
ok 100 - get_keys and as_hash
ok 101 - apply_dynamic
ok 102 - apply_static and print_config_def
ok 103 - get_abs_value() works when ratio_over references a floatOrPercent option
ok 104 - no interferences between DynamicConfig objects
ok 105 - apply dynamic over dynamic
ok 106 - extruder option is removed after normalize()
ok 107 - undefined extruder is populated with default extruder
ok 108 - defined extruder is not overwritten by default extruder
ok 109 - undefined solid infill extruder is populated with infill extruder
ok 110 - retract_layer_change is disabled with spiral_vase
ok
t/16_flow.t ..................... 
1..2
ok 1 - 'new_from_width' isa 'Slic3r::Flow'
ok 2 - 'new' isa 'Slic3r::Flow'
ok
t/17_boundingbox.t .............. 
1..5
ok 1 - 'new_from_points' isa 'Slic3r::Geometry::BoundingBox'
ok 2 - min_point
ok 3 - max_point
ok 4 - min_point equals to the only defined point
ok 5 - max_point equals to the only defined point
ok
t/18_motionplanner.t ............ 
1..22
ok 1 - An object of class 'Slic3r::MotionPlanner' isa 'Slic3r::MotionPlanner'
ok 2 - return path is valid
ok 3 - path length is greater than straight line
ok 4 - first path point coincides with initial point
ok 5 - last path point coincides with destination point
ok 6 - path is fully contained in expolygon
ok 7 - An object of class 'Slic3r::MotionPlanner' isa 'Slic3r::MotionPlanner'
ok 8 - return path is valid
ok 9 - path length is greater than straight line
ok 10 - first path point coincides with initial point
ok 11 - last path point coincides with destination point
ok 12 - path has no intersection with expolygon
ok 13 - An object of class 'Slic3r::MotionPlanner' isa 'Slic3r::MotionPlanner'
ok 14 - start point is contained in first expolygon
ok 15 - end point is contained in second expolygon
ok 16 - return path is valid
ok 17 - path length is greater than straight line
ok 18 - An object of class 'Slic3r::MotionPlanner' isa 'Slic3r::MotionPlanner'
ok 19 - start point is contained in second expolygon
ok 20 - end point is contained in first expolygon
ok 21 - return path is valid
ok 22 - path length is greater than straight line
ok
t/19_model.t ....................
1..4
ok 1 - An object of class 'Slic3r::Model::Object::Ref' isa 'Slic3r::Model::Object::Ref'
ok 2 - An object of class 'Slic3r::Pointf3::Ref' isa 'Slic3r::Pointf3::Ref'
ok 3 - origin_translation is modified by ref
ok 4 - layer_height_ranges roundtrip
ok
t/20_print.t ....................
1..5
ok 1 - An object of class 'Slic3r::Print' isa 'Slic3r::Print'
ok 2 - An object of class 'Slic3r::Config::Print::Ref' isa 'Slic3r::Config::Print::Ref'
ok 3 - An object of class 'Slic3r::Config::PrintObject::Ref' isa 'Slic3r::Config::PrintObject::Ref'
ok 4 - An object of class 'Slic3r::Config::PrintRegion::Ref' isa 'Slic3r::Config::PrintRegion::Ref'
ok 5 - An object of class 'Slic3r::GCode::PlaceholderParser::Ref' isa 'Slic3r::GCode::PlaceholderParser::Ref'
ok

Test Summary Report
-------------------
t/11_clipper.t                (Wstat: 512 Tests: 23 Failed: 2)
  Failed tests:  21, 23
  Non-zero exit status: 2
Files=19, Tests=456,  0 wallclock secs ( 0.10 usr  0.05 sys +  0.67 cusr  0.04 csys =  0.86 CPU)
Result: FAIL
Failed 1/19 test programs. 2/456 subtests failed.

````

arekm, 2015-06-17:
1.2.9:

````
ok 19 - intersection_ppl - result is not empty
ok 20 - intersection_pl - result is not empty
not ok 21 - intersection_pl - result is not empty

#   Failed test 'intersection_pl - result is not empty'
#   at t/11_clipper.t line 193.
#          got: '6'
#     expected: '5'
ok 22 - intersection_pl - result is not empty
not ok 23 - intersection_pl - result has same length as subject polyline

#   Failed test 'intersection_pl - result has same length as subject polyline'
#   at t/11_clipper.t line 206.
#          got: '92066651.9339821'
#     expected: '79586257.0728936'
# Looks like you failed 2 tests of 23.
Dubious, test returned 2 (wstat 512, 0x200)
Failed 2/23 subtests 
t/12_extrusionpathcollection.t .. 
1..18
ok 1 - 'collection object with items in constructor' isa 'Slic3r::ExtrusionPath::Collection'
ok 2 - no_sort is false by default
ok 3 - append ExtrusionPath::Collection
````

lordofhyphens, 2015-06-17:
Test suite failures are usually caught by the buildbot (
https://travis-ci.org/alexrj/Slic3r).

That being said, it still built, but the tests didn't all pass. It probably
still "works", but check the output.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2015-06-17:
Thank you @arekm, it doesn't fail on any of my Linux machines. Maybe it's specific to perl 5.20, I don't know.

arekm, 2015-06-17:
Maybe it is related to the fact that I'm using system libraries:
* admesh (http://github.com/admesh/admesh/ - 0.98.2)
* polyclipping (http://sourceforge.net/projects/polyclipping - 6.2.1)
* boost (boost.org, 1.58)
instead of bundled ones?

Polyclipping is first candidate probably.

alexrj, 2015-06-17:
Ah yes, that's most likely.  Slic3r uses Clipper (polyclipping) 6.2.9. Note that most of the bugs fixed in recent Clipper versions were reported by me (Slic3r), so it's very likely that our test case covers such issues.

Can you try to compile a fresh git clone of Slic3r so that we know that it works when using the expected versions?

arekm, 2015-06-17:
````
All tests successful.
Files=19, Tests=456,  0 wallclock secs ( 0.11 usr  0.03 sys +  0.65 cusr  0.07 csys =  0.86 CPU)
Result: PASS
````

Not good... since bundled libraries are not liked by Linux distributions. Closing anyway since works fine in "upstream".

closed 2015-06-17 by arekm

lordofhyphens, 2015-06-17:
Eh as long as it stays out of the main system hierarchy it's nbd. At least
for everyone except for the package maintainer.

alexrj, 2015-06-17:
6.2.9 is in the Clipper code repository. I know the issue with distros, but let's say that 6.2.9 is a dependency for Slic3r, so distros should cope with that and update their code.