slic3r/Slic3r #2906: Z banding fix

a4jp-com, 2015-06-12:
I have flashed my printer with Marlin firmware and fixed all the calculations but unfortunately my printer has the slightest ever z axis wobble. I even got the rod changed once and it is better than before but still not right. Is there anyway I can just tell slicer where the banding is? I have created anti shake parts for the printer and done as much as I can do except for keep on buying z rods until I get a perfectly straight one which is difficult. but it would be great if there was some g-code or a setting in slicer to fine tune the prints and get smooth walls.

peter1960, 2015-06-12:
Are the rods so bad they are actually moving the head?
Normally any z banding is due to the steps per mm not matching the thread pitch

alexrj, 2015-06-12:
I don't understand what are you asking. You should just fix your printer, what should Slic3r do?

whosawhatsis, 2015-06-12:
There are a lot of possible causes for these types of artifacts (see my article here: https://www.evernote.com/l/ANNwHDbE3dVGaaSClT2JJMcdHvmSmIKVSHw). Even if you knew the type, pitch and amplitude of these errors, the open-loop nature of the machines would make it virtually impossible to phase-align any type of software compensation for these mechanical issues.