slic3r/Slic3r #2907: 1.2.8 : zoom function can't be used in "3D" view mode after "2D" view mode.

asterixxxx, 2015-06-12:
OS: Win 8
version: 1.2.8

description: zoom function can't be used in "3D" view mode after "2D" view mode.

1-place an object
2-select "2D" view mode
3-switch back to "3D" view mode
4-bug: mouse wheel function "zoom in/out" will not work
5-workaround fix:: select "preview" tab and then "3D" tab again.

xoan, 2015-06-13:
Not reproducible un GNU/Linux.

    ~$ uname -a
    Linux meikian 3.2.0-4-686-pae #1 SMP Debian 3.2.63-2+deb7u1 i686 GNU/Linux
    ~$ RepRap/Software/Slic3r-1.2.8/bin/slic3r --version
    1.2.8

phizz166, 2015-06-13:
No problems for me on Win 8.1 / 1.2.8 following the same steps. Works with both mouse wheel scrolling and laptop trackpad scrolling.

krizzap, 2015-07-05:
I can confirm this issue is also present in Win7 running the x64 version of 1.2.9

andyjbm, 2017-10-05:
Using 1.30-dev and still a problem. All you have to do is select a dropdown on the right hand side pane and once the 3d view has lost focus the scroll wheel no longer zooms. The only way it would appear to get the focus back to 3d view is to click on the preview tab and then back to the 3d view tab.

A.

andyjbm, 2017-10-05:
Sorry - windows 7 x64 Slic3r 1.3-dev

lordofhyphens, 2017-10-05:
@andyjbm did you click back in the 3d view? It's normal behavior for while focus off of the 3d view for it to not respond to the scroll wheel.

lordofhyphens, 2017-10-05:
@andyjbm I can't reproduce on my win7 system either.

closed 2017-10-05 by lordofhyphens

andyjbm, 2017-11-26:
Hmmm, Sorry didn't receive the email notifications of the replies to my post.

yes @lordofhyphens I did click back in 3d view. I could pan and rotate holding down left mouse button but could not zoom using the mouse wheel without performing the above.

But whaddaya know its working now. What was happening when you moved the mouse wheel was that which ever dropdown box had focus was going mental scrolling,  and clicking in 3d view would not return focus from the dd box to 3d view. But I could pan/rotate with left mouse button and drag objects around the platter.

I'll pay attention if it happens again I'll see what work I've got loaded...

Andy.

lordofhyphens, 2017-11-27:
Np. @andyjbm, if you can catch it in the act it and provide specific reproduction steps (make the issues stand up and bark on command, as it were), I'll gladly reopen and we can try to figure out what is going on (especially with a build from https://dl.slic3r.org/dev/win).

lordofhyphens, 2017-11-27:
Since this was originally against 1.2.8, quite a bit "under the hood" has changed w/r/t libraries, etc. Especially with the download builds that are provided.

dartrax, 2017-12-19:
I think this might be related to #4217 (Mouse wheel zoom does not work in 3D-Tab after Drag&Drop-Operation).

andyjbm, 2017-12-20:
Hi,

Well I did it again. 

To reproduce. 

Drag an object onto the platter.

Make sure you can zoom with the mouse wheel.

Hit the Delete All button.

Zoom is now broken and the selection boxes on the rhs panel go mad when you scroll the mousewheel.

V1.3.0-dev

Regards,

Andy.

dartrax, 2018-01-22:
Hi andyjbm,
I can reproduce that exactly and described your steps to reproduce in #4217.

andyjbm, 2018-01-23:
Hi dartrax,

I found some more significant info towards recreating this issue and posted to #4217.

A.