slic3r/Slic3r #2908: Request: manual color settings for toolpath previews

phizz166, 2015-06-13:
Currently in 2D and 3D preview mode the software shows infill, perimeter and support toolpaths as different colors. It would be great to be able to customize these colors, and have additional separate color settings for solid infill, unsupported bridges, and support interface layers. This distinction would help a lot in evaluating a job prior to printing.

In addition, with the new filament color feature (1.2.8), the user could set any of the toolpaths to [use filament color] instead of a fixed value. This would be a helpful reminder for people with a lot of different filaments, and it would be especially useful for those of us with multiple extruders, showing exactly which parts of the model will print with which tool.

Allowing the user to customize the color of the viewport background and grid might also be necessary for readability.

Finally, in the 3D view it'd be nice to have a toggle for perspective/isometric view. Perspective just looks nicer :)

a4jp-com, 2015-06-13:
+1 for the color idea
+1 for perspective

Patola, 2017-01-01:
Hi, is that issue still active? I would like to second this setting, and add that separate colors for dual or multiple extrusion is a complete *must*. When you use multiple extrusion, you *have* to know which extruder is doing what, but slic3r's preview windows combine them in one single color.

foreachthing, 2017-04-07:
This was bothering me as well!
So, I made [this](https://github.com/foreachthing/Slic3rPostProcessing) to (re-) view the CGode in CraftWare.

* before:
[image]
* after:
[image]