slic3r/Slic3r #2909: Aligned or nearest seam position in overhang points when a non overhang point is available

xoan, 2015-06-13:
Not sure if it is reproducible in all cases, but in 45º overhang (like teardrop holes), a pair of the nearest overhang corners are choosed as start point for preimeters.

STL file: [`x-end-idler.stl`](https://github.com/josefprusa/Prusa3-vanilla/blob/master/distribution/x-end-idler.stl)
Configuration file: [`FoldaRap - 1.2.8.ini`](https://gist.github.com/xoan/314f57d2cd5255d42d44)

[image]

This layer is one of the layers where the belt hole is near to be closed, and it has a 45º overhang inside, splitted by a stress relief that made the bug more noticeable.

This behaviour makes that the plastic tends to accumulate at this overhang points, so curling up and some similar faults appear more easy.

xoan, 2015-06-13:
As the [manual](http://manual.slic3r.org/expert-mode/fighting-ooze) says:

* `Nearest` - This will try to choose a concave non-overhang vertex so that the seam is hidden in the inside of the concave angle. If no concave non-overhang vertices are available, it will choose a convex non-overhang vertex. If none are available, it will choose a non-overhang vertex. The choice among the candidates is operated so that the start point is the nearest to the previous extruder position. So this option will optimize for short travels.
* `Aligned` - This will use the same logic as `Nearest` for finding the candidates, but it will choose the one that is nearest to the start point of the previous layer. This will ensure the seam is mostly aligned throughout the whole object.

alexrj, 2015-06-13:
Good catch.   The overhang logic is not actually applied to *Nearest*, while it is to *Aligned* but it's a bit conservative and it's not considering those vertices as overhangs (overhang perimeters are the ones where we apply bridge speed etc.).  But I think here we should apply a different logic.

xoan, 2015-06-13:
Thanks for considering that! I have no active layer fan, and I print almost in PLA, which is very curling+up friendly :\

> (overhang perimeters are the ones where we apply bridge speed etc.).

I suposed something like this. 

> But I think here we should apply a different logic.

I think for the seam position, any wall angle greater than 0º should be considered overhang (in most of the islands, there is almost allways another corner with non-overhang :)

alexrj, 2015-06-14:
I'll park this issue for a bit, as I started implementing some alternative algorithm but it looks there are logical problems. Since you're not using the *External Perimeters First* option, the inner loops are extruded first. And none of their vertices are overhangs in your case as they're well contained in the lower layers...  I think that adding a fan is the proper solution.

xoan, 2015-06-14:
OK, I understand the complex of this logic. I'll wait for it :)

I'd installed a fan some time ago, but in most of this cases, curling up due plastic accumulation was appear anyway. In fact, following the same STL as in the first message, the other overhang vertices are printed well with and without fan.

About `External perimeters first` option, when overhang is propertly detected (vertices with concurrent _bridged_ perimeters), outer perimeter is started in another vertice, and this current implemented behaviour reduces a lot the plastic tension.

xoan, 2015-06-14:
> when overhang is propertly detected (vertices with concurrent bridged perimeters), outer perimeter is started in another vertice, and this current implemented behaviour reduces a lot the plastic tension.

This happens only when `Aligned` seam position, not when `Nearest`.

Just another test case:

STL file: [`y-idler`](https://github.com/josefprusa/Prusa3-vanilla/blob/master/distribution/y-idler.stl)
Configuration file: [`FoldaRap - 1.2.8.ini`](https://gist.github.com/xoan/314f57d2cd5255d42d44)

[image]

[image]

In this RP part, the effect is not so big and not so important, but it is noticeable by comparing both sides of the hole (this part was printed within a plate, so seam position is not in the seam vertice than G-code preview, but it is in the other one similar point)

xoan, 2017-03-21:
I'm going to close this: I think it is a duplicated (it is closely related to) of #2818

closed 2017-03-21 by xoan