slic3r/Slic3r #2910: Perimeter pieces stick out in different directions.

lonelymyp, 2015-06-13:
[image]

Config:
https://drive.google.com/file/d/0B2sGop41unyOZmdQSE5BTHVETEE/view?usp=sharing
STL: 
https://drive.google.com/file/d/0B2sGop41unyOTWN2UXpxNFljVVU/view?usp=sharing

alexrj, 2015-06-13:
Does Slic3r's 3D preview show those?

alexrj, 2015-06-13:
Does they show also in any 2D preview like Slic3r's one, Pronterface or http://gcode.ws ?

lonelymyp, 2015-06-13:
Yes, the same.
[image]

lonelymyp, 2015-06-13:
G-code:
https://drive.google.com/file/d/0B2sGop41unyOS1RMQ3ZsbHNtVDA/view?usp=sharing

alexrj, 2015-06-14:
Okay, this is Slic3r trying to generate the thin walls for some of the teeth:

[image]

alexrj, 2015-06-15:
I fixed this. However I'd like to do more work on it.

closed 2015-06-20 by alexrj